MrChromebox/firmware #518: SD card not recognized on startup

Keylessboi, 2023-09-18:
Upon startup my Nintendo SD card 128 gb SD card isn’t recognized so I cannot boot to it. Here is some device specs and a further explanation: 
Board: CANDY 
Firmware version is: Mrchromebox-4.20.1 07/21/2023
Model: Dell Chromebook 11 3120 OS: Linux mint (efi partition is on emmc and all files are on sd card inserted via the sd card reader on the Chromebook) 
Issue: I have to reboot many times (anywhere to ~3-40) before the SD card is recognized. When it is not recognized I just get redirected to the grub terminal. Everything works just fine when the sd is recognized. 
FULL ROM: Yes 

The layout is
eMMC:
500MB EFI at /boot/EFI
SD:
128GB EXT4 at /
I wasn't able to use the boot from file option to find the drive either.
I also tried putting everything on the SD card where it would just go to efi shell.
The GRUB binary is loaded from the eMMC and when it tries to load the config off the SD Card it fails most of the time as if it didn't exist and takes multiple reboots until GRUB and EDK2 actually see it

I’m happy to share any extra details.

MrChromebox, 2023-09-18:
on older devices like CANDY, there's nothing I can do debug these types of problems, as they have no ability to output serial debug logging which is needed to troubleshoot the issue. 

For USB detection, boot to the UEFI main setting screen, remove/reinsert the USB device, wait 3s, then check the boot menu. Other than that, all you can do is find USB media that is less temperamental

Keylessboi, 2023-09-18:
Is there another device you know of I can run IntelliJ on that is extremely cheap like some of the candy devices?

MrChromebox, 2023-09-18:
I really don't keep up with used Chromebook prices. The chrultrabook discord server will provide better insight there

hede5562, 2026-11-04:
As a workaround, the following is working for my Coral based Chromebook, which also suffers from the effect of not having the SD card present after reboot:

https://issei.space/blog/linux-chromebook-sd-card-on-boot/

> 
> \# /etc/systemd/system/rescan-sd-card.service
> [Unit]
> Description=Remove and rescan MMC and PCI devices before mounting filesystems
> DefaultDependencies=no
> Before=local-fs.target
> 
> [Service]
> Type=oneshot
> ExecStart=/bin/sh -c 'echo 1 > /sys/class/mmc_host/mmc0/device/remove'
> ExecStartPost=/bin/sh -c 'echo 1 > /sys/class/pci_bus/0000:00/rescan'
> RemainAfterExit=yes
> 
> [Install]
> WantedBy=local-fs.target

`# systemctl enable rescan-sd-card.service`

Please keep in mind that "mmc0" and  pci bus number 0 may not be generic and may vary.

MrChromebox, 2026-11-05:
Confirmed the same intermittent SD detection on **Swanky** (same rambi/Bay Trail SD host as Candy: SoC `pci 12.0`, identical SD3 GPIO/caps). Behavior is reproducible but inconsistent — card sometimes missing from the firmware boot path until a reboot lucks out — so this looks like a shared BYT/edk2 SD enumeration quirk rather than a Candy-specific board bug.

Still no practical serial debug on these devices; leaving open as a known platform limitation unless someone finds a workable SD host init/timing change to try.