MrChromebox/firmware #521: asus chromebook

tfc31, 2023-09-23:


tfc31, 2023-09-23:
bonjour j'ai suivi les étape voila se que j'ai comme le dispositif error et même j'ai rentrer dans recovery utility pour faire pareille je peux même pas restaurer a l'état d'usine et j'ai regarder sur YouTube d' autre marque comme hp il faut retirer un vis de sécurité sur la carte mère merci d'avance
[image]
[image]
[image]

MrChromebox, 2023-09-23:
I don't support running Windows on these devices -- coolstar does.

I also don't speak French, and putting your messages thru Google translate isn't ideal.

I don't even know what device (board name) you have, because you cut it off in your screenshot.

closed 2023-09-23 by MrChromebox