MrChromebox/firmware #523: I am not able to disable WP

H644b, 2023-09-24:
Hi I have a Lenovo Chromebook 300e 2nd gen and I took out the battery and did the commands for disabling software write protection it is not letting me disable it. Can someone help me? btw my chromebook is an phaser360 and octopus

H644b, 2023-09-24:
I was also not able to disable hardware write protection

MrChromebox, 2023-09-24:
where did you get the idea you needed to run anything to disable software write protect? You do not and should not.

Boot the device with the internal battery disconnected, and run the script. that's it.

closed 2023-09-24 by MrChromebox

H644b, 2023-09-24:
When I run a raw sh1m it says I have hardware write protect enabled

H644b, 2023-09-24:
Also when I run your Kodi script it says I have write protection enabled

H644b, 2023-09-24:
Also when I try to set flags in says this: Calibrating delay loop... OK. coreboot table found at 0x79b29000 read_from_file failed to read 2641920 bytes No fmap entries found in /tmp/tmp.BoY73dB3aN Block protection could not be disabled! Reading flash... SUCCESS Setting GBB flags from flags: 0x00000000 to 0x8090..successfully saved new image to: /tmp/tmp/BoY73dB3aN WARNING: System GBB Flags are NOT changed!!! ERROR: You must disable write protection before setting flags.

MrChromebox, 2023-09-25:
because hardware WP is enabled. I don't understand the disconnect here.

H644b, 2023-09-25:
I do not understand since I unplugged the battery and ran off the charger.

MrChromebox, 2023-09-25:
take a screenshot of the script menu. Also provide the output from `sudo crossystem wpsw_cur`

H644b, 2023-09-25:
I am not actually booted into chrome os im actually in the shell from sh1mmer

H644b, 2023-09-25:
but I can still give you the results by booting into E-Halycon

MrChromebox, 2023-09-25:
I have nothing to do with and don't support shimmer, sorry.

H644b, 2023-09-25:
No but in general I just need to know why the WP is not being turned off even after I disconnected the battery

H644b, 2023-09-25:
[image]
[image]

H644b, 2023-09-25:
Also does the fact that I sent my chromebook for a repair a month ago change anything like could they have debugged it and permanently set WP to be on?

MrChromebox, 2023-09-25:
you disconnected the battery cable from the mainboard, not the battery, right? if you disconnect from the battery it won't work

> Also does the fact that I sent my chromebook for a repair a month ago change anything like could they have debugged it and permanently set WP to be on?

lol, no

H644b, 2023-09-25:
Yes I disconnected the battery cable from the mainboard.

MrChromebox, 2023-09-25:
is it a managed device?

H644b, 2023-09-25:
Yes

MrChromebox, 2023-09-25:
then that's why

H644b, 2023-09-25:
Oh so you mean that they can change wether you have it on or off permanently?

MrChromebox, 2023-09-25:
not permanently. They (Goole) simply updated the CR50/GSC firmware to not disable FW WP on battery disconnect if the device is managed

H644b, 2023-09-25:
Is their a way to downgrade firmware? I tried using a raw shim.

H644b, 2023-09-25:
Or do I have to get a debug cable or dig in the mainboard

MrChromebox, 2023-09-25:
the debug cable is also locked out when the device is managed

H644b, 2023-09-25:
What about the Rom Chip would there be a way of doing something with that to flash firmware on it?

MrChromebox, 2023-09-25:
<shrugs>

H644b, 2024-02-10:
Epilogue: I was able to use the pencil method and now I turned off WP.

H644b, 2024-02-10:
: D