MrChromebox/firmware #528: Lenovo C13 Yoga with 4K screen's touchscreen does not work on fullrom

junnikokuki, 2023-10-11:
The C13 Yoga with 4k screen's touchscreen does not work. There is not touchscreen related hardware. I reinstalled dualboot windows and the touchscreen works ok.
I have exported the dsdt and ssdt of the dualboot windows. I can send to you if you need.

MrChromebox, 2023-10-11:
the dsdt and ssdt are not useful but thanks. Does the touchscreen work under Linux with the Full ROM firmware? Can you provide a cbmem log from it?

junnikokuki, 2023-10-11:
[cbmem-c13.log](https://github.com/MrChromebox/firmware/files/12872482/cbmem-c13.log)

A cbmem log from Live ubuntu cd with try ubuntu. No touchscreen, trackpoint and touchpad work. If it not helps, I can install a ubuntu to test.

MrChromebox, 2023-10-11:
it looks like the touchscreen is failing detection for some reason, so the ACPI isn't getting added. Since morphius only has one touchscreen option, and no SKUs lack a touchscreen, I can probably drop the detection and just assume it's present

junnikokuki, 2023-10-12:
Change generic.detect to generic.probe or just remove this flag?

MrChromebox, 2023-10-12:
just comment out or remove. the 'probed' flag has no function outside of the ChromeOS kernel - it's not recognized under upstream Linux or Windows

junnikokuki, 2023-10-12:
OK, I'll try it.

junnikokuki, 2023-10-12:
Hi, could you compile the bios for me? I checked out the 4.21.0 tag and build with no modification, but the bios won't boot(the power light is on but the screen is black), if I flashed back to official 4.21.0 20231001, it boots normally.

MrChromebox, 2023-10-12:
yes, you cannot build for Zork boards from the 4.21 branch, they require a hack to work around some vboot issues which I didn't feel like forking the module for

MrChromebox, 2023-10-12:
try this build please

[coreboot_edk2-morphius-mrchromebox_20231012.zip](https://github.com/MrChromebox/firmware/files/12886591/coreboot_edk2-morphius-mrchromebox_20231012.zip)

junnikokuki, 2023-10-13:
Fixed! Thank you. I'll test other functions.

MrChromebox, 2023-10-13:
fixed build pushed to the main script, closing

closed 2023-10-13 by MrChromebox