MrChromebox/firmware #529: Wake up failure after suspend / sleep / hibernation.

GamePlayer-8, 2023-10-14:
Good afternoon,

I've got a problem with a custom Full ROM `coreboot_edk2-barla-mrchromebox_20230930.rom` on `HP Chromebook 11A G6 EE`.

Every time I'm trying to suspend / sleep / hibernate the system, Chromebook no longer reacts on Power / Lid button when I want to wake it up.

After pressing Power button for 30s~ I can use the Chromebook again, but it boots up from the beginning.

MrChromebox, 2023-10-14:
known issue, is being investigated

Sma1525, 2023-10-15:
Hey MrChomebox,
in case it helps having the same experience with HP x360 G3 EE (Vortininja)
Thanks!

MrChromebox, 2023-10-15:
@Sma1525 that's a completely different device/platform, please ask for help on Discord to confirm it's a firmware issue before opening an issue here.

@GamePlayer-8 issue on BARLA has been fixed and new firmware pushed

closed 2023-10-15 by MrChromebox

GamePlayer-8, 2023-10-15:
Confirmed, on the latest ROM it works.