MrChromebox/firmware #530: Uncontrolled underclocking on AMD CPUs / Chromebook.

GamePlayer-8, 2023-10-15:
Good evening,

There's an issue with AMD CPU `A4-9120C` in `HP Chromebook 11A G6 EE`, what's having a default `1.2 Ghz - 1.8 Ghz` clock speed, with boost up to `2.4 Ghz`.

Unfortunately with the latest Full ROM & older ones, CPU likes to underclock itself at the beginning of chromebook startup. The exact underclock frequency is `798 Mhz` and it's mostly static.

Driver used by the Linux kernel for CPU Frequency: `auto-cpufreq`.
Kernel: `6.5.5`

I've tried adding `intel_pstate=disable amd_freq_sensitivity=off` to the kernel boot params, but didn't helped.
Underclock is noticable before kernel & bootloader starts.

Only hard reset on battery power makes the CPU work normally although it might be random.

MrChromebox, 2023-10-16:
what exactly are you using to determine this?

My Stoney Chromebook here running Pop_OS! is idling between 1.8-2.0GHz per /proc/cpuinfo

This sounds like a kernel issue not a firmware one.

GamePlayer-8, 2023-10-16:
> what exactly are you using to determine this?

I've used `cpupower` / `htop` / `top` or whatever other tool what's reading `/proc/cpuinfo`.

I've tried with Debian / Ubuntu / Artix / Arch kernels and I'm getting the same issue.

MrChromebox, 2023-10-16:
Windows is reporting normal CPU speeds as well here.  What version of the firmware are you running?

MrChromebox, 2023-10-16:
powertop doesn't even report 800MHz in the freq stats for me - only 1200/1400/1600 so not sure how accurate that is since not even listing the turbo boost states

GamePlayer-8, 2023-10-16:
FW: `MrChromebox-4.21.0-8-geb419016267 (10/15/2023)`

`cpupower` also doesn't report 800 Mhz as a supported frequency. Atm I've hard reseted chromebook to escape from underclocking. `dmesg` doesn't report anything related to CPU clock neither.

GamePlayer-8, 2023-10-16:
38°C - 65°C

MrChromebox, 2023-10-16:
I'm at a loss, I can't reproduce and can't find anyone else with the issue either

GamePlayer-8, 2023-10-17:
This time I've got `dmesg` errors:
```log4j
[12833.148139] clocksource: timekeeping watchdog on CPU1: Marking clocksource 'tsc' as unstable because the skew is too large:
[12833.148154] clocksource:                       'hpet' wd_nsec: 505890273 wd_now: 89e44a wd_last: 1b5da6 mask: ffffffff
[12833.148160] clocksource:                       'tsc' cs_nsec: 506839844 cs_now: 8f002d02 cs_last: 5ec157ad mask: ffffffffffffffff
[12833.148164] clocksource:                       Clocksource 'tsc' skewed 949571 ns (0 ms) over watchdog 'hpet' interval of 505890273 ns (505 ms)
[12833.148168] clocksource:                       'tsc' is current clocksource.
[12833.148185] tsc: Marking TSC unstable due to clocksource watchdog
[12833.148213] TSC found unstable after boot, most likely due to broken BIOS. Use 'tsc=unstable'.
[12833.148214] sched_clock: Marking unstable (12833148972745, -1325832)<-(12833150600200, -2387436)
[12833.147825] clocksource: Checking clocksource tsc synchronization from CPU 0 to CPUs 1.
[12833.147852] clocksource: Switched to clocksource hpet
```

Not sure at this point if the problem is about the kernel too.

Juma7C9, 2023-10-23:
Can confirm, I'm having the same issue with more than one distribution (tried OpenSuse Tumbleweed and Debian 12), with both default kernels and the custom kernels provided by the Chrultrabook project.

I also tried to downgrade the firmware to each version down to the July version, but to no avail.

To exclude an issue with the frequency reporting I tried a simple `zstd` compression benchmark on both Linux and ChromeOS (reverting to the default firmware/image), and the ChromeOS is in fact ~5-6 times faster, all other things equal.

One of the consistently reported difference is on the `lscpu` output, where `CPU(s) scaling MHz` is stuck on 50% on UEFI Linux, and above 100% on ChromeOS.

As I'm not sure what else to test, if there are suggestions I may debug further.

coolstar, 2023-10-27:
bulldozer throttles at 60 C. Repaste the CPU and it'll clock higher

GamePlayer-8, 2023-10-28:
mine does at whatever temperature. Thermal paste is new on the CPU.

GamePlayer-8, 2023-11-03:
Is there any way to forcefully disable thermal protection and everything related to that in the firmware to check if that may cause the issue?

GamePlayer-8, 2023-12-05:
Adding `mitigations=off` to the kernel parameters seems to have fixed the issue although it exposes multiple vulnerabilities on the way.

`lscpu` output after rebooting the system:

```shell
Architecture:            x86_64
  CPU op-mode(s):        32-bit, 64-bit
  Address sizes:         48 bits physical, 48 bits virtual
  Byte Order:            Little Endian
CPU(s):                  2
  On-line CPU(s) list:   0,1
Vendor ID:               AuthenticAMD
  Model name:            AMD A4-9120C RADEON R4, 5 COMPUTE CORES 2C+3G
    CPU family:          21
    Model:               112
    Thread(s) per core:  1
    Core(s) per socket:  2
    Socket(s):           1
    Stepping:            0
    Frequency boost:     enabled
    CPU(s) scaling MHz:  119%
    CPU max MHz:         1600,0000
    CPU min MHz:         1200,0000
    BogoMIPS:            3195,13
    Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mc
                         a cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall n
                         x mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_go
                         od acc_power nopl nonstop_tsc cpuid extd_apicid aperfmp
                         erf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 
                         movbe popcnt aes xsave avx f16c lahf_lm cmp_legacy svm 
                         extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch 
                         osvw ibs xop skinit wdt lwp fma4 tce nodeid_msr tbm per
                         fctr_core perfctr_nb bpext ptsc mwaitx cpb hw_pstate ss
                         bd vmmcall fsgsbase bmi1 avx2 smep bmi2 xsaveopt arat n
                         pt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushby
                         asid decodeassists pausefilter pfthreshold avic v_vmsav
                         e_vmload vgif overflow_recov
Virtualization features: 
  Virtualization:        AMD-V
Caches (sum of all):     
  L1d:                   64 KiB (2 instances)
  L1i:                   128 KiB (2 instances)
  L2:                    2 MiB (2 instances)
NUMA:                    
  NUMA node(s):          1
  NUMA node0 CPU(s):     0,1
Vulnerabilities:         
  Gather data sampling:  Not affected
  Itlb multihit:         Not affected
  L1tf:                  Not affected
  Mds:                   Not affected
  Meltdown:              Not affected
  Mmio stale data:       Not affected
  Retbleed:              Vulnerable
  Spec rstack overflow:  Not affected
  Spec store bypass:     Vulnerable
  Spectre v1:            Vulnerable: __user pointer sanitization and usercopy ba
                         rriers only; no swapgs barriers
  Spectre v2:            Vulnerable, STIBP: disabled, PBRSB-eIBRS: Not affected
  Srbds:                 Not affected
  Tsx async abort:       Not affected
```

jonas5, 2023-12-14:
> mine does at whatever temperature. Thermal paste is new on the CPU.

T

> bulldozer throttles at 60 C. Repaste the CPU and it'll clock higher

I had this exact same issue. But a different CPU architecture.

My Lenovo c13 yoga did only do 400mhz on each core, I tried to roll back into stock firmware it still did some weird CPU throttling. I was sure it was some bios issue not able to scale back from power saving mode.
In last attempt I had to open the laptop to disconnect the laptop to enable WP. I left the lid open and the throttling was gone. Regardless firmware used. 

Repasted tfe CPU as it was stale paste on CPU.
Now it all works well, 
Guess what I'm trying to say - thank you!

Stefan-make, 2024-01-18:
I have this problem too.
I can cause it by hard reset "refresh+power", after that:
`stefan@debian:~$ cat /proc/cpuinfo | grep Hz
cpu MHz		: 798.523
cpu MHz		: 798.498
stefan@debian:~$ cat /proc/cpuinfo | grep Hz
cpu MHz		: 798.507
cpu MHz		: 798.514
stefan@debian:~$ cat /proc/cpuinfo | grep Hz
cpu MHz		: 798.536
cpu MHz		: 798.504
`
Reboot dont help, but to fix it you must shutdown and wait 10-15sec.
After that:
`stefan@debian:~$ cat /proc/cpuinfo | grep Hz
cpu MHz		: 2183.506
cpu MHz		: 2178.905
stefan@debian:~$ cat /proc/cpuinfo | grep Hz
cpu MHz		: 2395.613
cpu MHz		: 2395.610
stefan@debian:~$ cat /proc/cpuinfo | grep Hz
cpu MHz		: 2395.633
cpu MHz		: 2395.572
stefan@debian:~$ cat /proc/cpuinfo | grep Hz
cpu MHz		: 1197.809
cpu MHz		: 1197.809
`

jonas5, 2024-01-22:
These Chromebooks aren't really designed for heavy duty tasks such as gaming. 
The cooling solution is minimal and over short amount of time it may overheat and start throttling. 

I have tried many different kernels and variations of flavoured operating systems seem to not change the outcome. 

Other vendors have been doing some bios update to resolve the switching back to normal operational mode happen. For now using refresh+power button did work for me too. It only works temporarily until next time it overheats, but it is easier than removing the battery.

ccordero90, 2024-07-24:
I also have a barla with the exact same issue. I am running ` Fw Ver: MrChromebox-4.21.1`  on archlinux `Kernel: 6.6.16-stoney ` `CPU: AMD A4-9120C RADEON R4 2C+3G (2)`

doing `systemctl poweroff` and waiting 15 secs resolves the issue

HarderLemonade, 2024-08-01:
I also have a G6 11A EE. I have been pulling my hair out, building custom kernels, installing debian sid, stable, oldstable, oldoldstable... thinking it was a kernel issue. At one point I also thought it was solved by disabling CPU mitigations, or compiling the kernel without them. I compiled a kernel with all intel eist / p4 clockmod (any intel power drivers) and intel microcode removed. I was almost sure this did the trick and benchmarks confirmed it. I was careful to not change the kernel or any firmware versions. And my system kept freaking slowing down to quite honestly an unusable miserable speed. I use lightweight DEs, XFCE or LXDE. Sometimes I even use Seamonkey instead of Firefox, when I can anyways. And this laptop was still miserable to use.

The posts above are correct though. If you power the machine OFF completely, let it sit a minute (>15 seconds, as stated), and power it on by opening the lid, it will boot up with proper CPU scaling.

I used openarena as a performance test, since it is in all debian versions going back to kernel 4.x even, and when this bug occurs it jitters quite a bit, even in 640x480 windowed all low settings. However, shut the box down, leave off 30 seconds & open lid, and it will run the game @ 1366x768 32 bit all settings high or maxed out, bloom and flare on, even 2x antialiasing and it runs at what looks to be a silky smooth 60 FPS even with the governor set to "powersave" 

Since I've had this problem on kernels ranging from 4x - current 6.6, wouldn't this be a firmware issue??  Thank you for all your work, it is incredibly awesome that you've enabled so many of these machines to avoid their destiny in the landfills. You seriously rock, and I have a ton of respect Mr Chromebox. Thank you for doing what you do.

MrChromebox, 2024-08-01:
so firmware wise, there hasn't been much change to Stoneyridge in quite some time, though there was a change to the SMU (power mgmt) firmware back in Oct 2023. So this change likely would have affected the 4.22 firmware as well as the current 2405 firmware. If someone wants to test a previous release (4.20 has the old SMU firmware) we can rule that in/out pretty easily. Or I can do a test build of 2405 with the old SMU firmware

HarderLemonade, 2024-08-01:
I'll gladly try the previous firmware, would love to be of any help. Can you point me towards it? I installed what I'm using now /w your script from inside Chrome OS.

MrChromebox, 2024-08-01:
drop me an email or ping me on Discord and I'll sort you out to flash an older build / test version

ccordero90, 2024-08-08:
i would like to try the older firmware as well

MrChromebox, 2024-08-08:
> i would like to try the older firmware as well

there's no reason for multiple people to try the same thing, especially when it didn't work for the first person

noyz143, 2025-08-09:
@GamePlayer-8  i have the same problem, may i ask if you resolve this problem, hp 11 G6 EE with processor amd a4 9120c.  im running it with windows 10 operating system.

noyz143, 2025-08-09:
im stuck @ 0.79 ghz at startup.

noyz143, 2025-08-11:
@GamePlayer-8  still the same.

HarderLemonade, 2025-08-15:
You need to power off the machine, and boot it from cold. Under linux, disable sleep. I tried to diagnose the problem with Mr Chromebox a while back, we were unable to figure out what exactly was going on... But, the problem only happens when resuming from suspend. Boot from cold (off) and you should be fine, if you never go into suspend.

noyz143, 2025-10-22:
> as of latest firmware I am experiencing the same issue
> 
> as a workaround I have disabled sleep completely:
> 
> ```
> /etc/systemd/sleep.conf.d/disable-sleep.conf
> 
> [Sleep]
> AllowSuspend=no
> AllowHibernation=no
> AllowHybridSleep=no
> AllowSuspendThenHibernate=no
> ```
> 
> `$ sudo systemctl mask sleep.target suspend.target hibernate.target hybrid-sleep.target`

noyz143, 2025-10-23:
how to do it on windows?

MrChromebox, 2026-02-06:
> [@MrChromebox](https://github.com/MrChromebox) Sorry to bother you, but I want to ask if this problem has been finally solved? HP11A G8ee a4-9120C

sorry no idea, and I don't have one of these devices for testing

mehdilight, 2026-09-17:
For anyone still troubleshooting this on Stoney Ridge APUs (A4-9120C / `barla` / `careena`), here is the technical root cause and a verified breakdown of why the cold shutdown workaround works:

### Root Cause: Hardware SMU Clamp in MSR `0xC0010061`
Inspecting the AMD APU hardware registers via `/dev/cpu/0/msr` reveals the exact mechanism:
- **Throttled state:** AMD Model-Specific Register `0xC0010061` (`PStateCurLim`) reads `0x33` (`CurPstateLimit = 3`). The on-die SMU forces a hard clamp to P-state 3 (~798 MHz). Software governors like `cpupower` or `auto-cpufreq` cannot override this because the MSR is read-only at the hardware level.
- **Unthrottled state:** When operating normally, `0xC0010061` reads `0x30` (`CurPstateLimit = 0`), allowing full scaling up to base (1.6 GHz) and boost Pb0 (2.40 GHz).

### Why `reboot` Fails vs. Why Cold `poweroff` Works
- A software `reboot` (warm restart) **does not clear the clamp** because motherboard standby power rails remain energized. The SMU never power-cycles and retains the throttled state across reboots.
- Doing a complete shutdown (`poweroff`) and **waiting 15–20 seconds** allows motherboard capacitors to fully discharge. When turned back on with the charger connected, the SMU cold-initializes from scratch, detects full AC power, sets `CurPstateLimit = 0`, and the CPU immediately scales to 2.40 GHz.

### How to Check Your Register (run as root / with sudo):
```bash
sudo modprobe msr 2>/dev/null; sudo python3 -c "import struct, os; fd=os.open('/dev/cpu/0/msr', os.O_RDONLY); os.lseek(fd, 0xC0010061, 0); val=struct.unpack('<Q', os.read(fd, 8))[0]; print(f'MSR 0xC0010061: 0x{val:X} | CurPstateLimit: {val & 0x7}')"
```
- If it prints `CurPstateLimit: 3`: The APU is hardware-clamped to 798 MHz.
- If it prints `CurPstateLimit: 0`: Full 2.40 GHz boost is unlocked and active.

Tested and verified on HP Chromebook 11A G6 EE (`barla` rev6) running Alpine Linux Edge with MrChromebox UEFI Full ROM. Under `perf stat`, both cores sustain ~2.39 GHz under load at an ice-cold 43°C.

MrChromebox, 2026-09-17:
flash this and see if it fixes the SMU issue. cold poweroff after flashing plz

[coreboot_edk2-careena-mrchromebox_20260917.zip](https://github.com/user-attachments/files/32320323/coreboot_edk2-careena-mrchromebox_20260917.zip)

mehdilight, 2026-09-17:
> flash this and see if it fixes the SMU issue. cold poweroff after flashing plz
> 
> [coreboot_edk2-careena-mrchromebox_20260917.zip](https://github.com/user-attachments/files/32320323/coreboot_edk2-careena-mrchromebox_20260917.zip)

@MrChromebox Flashed `coreboot_edk2-careena-mrchromebox_20260917.rom` and tested on HP Chromebook 11A G6 EE (`barla` rev6).

The issue is completely resolved.

Here are the test results while running on battery power (62%):
1. Cold boot: `MSR 0xC0010061` reads `0x30` (`CurPstateLimit: 0`), scaling to 2.40 GHz immediately.
2. Warm software reboot (`reboot`): On the previous firmware, a warm reboot on battery always clamped the APU to 798 MHz (`CurPstateLimit: 3`). With this test build, it booted straight back into `CurPstateLimit: 0` at 2.40 GHz.
3. Hard EC reset (`Refresh [F3] + Power`): On the previous firmware, an EC reset guaranteed the 798 MHz lock. With this build, it rebooted in 6 seconds and remained completely unlocked (`CurPstateLimit: 0` at ~2.395 GHz under full load via `perf stat`).

The 15-20 second shutdown wait and AC connection workarounds are no longer needed. Thank you for this build.

GamePlayer-8, 2026-09-17:
Tested as well and after a cold boot, I put the CPU under maximum load. Throttles at 66 - 67°C from 2.4 Ghz to 1.6~1.8 Ghz, but doesn't seem to go down to 800 Mhz. will see how the CPU behaves in the long run; the issue seems to be resolved for now.

MrChromebox, 2026-09-17:
@mehdilight (or @GamePlayer-8) mind testing some builds for me to see if we can isolate what's causing the issue?

[coreboot_edk2-careena-mrchromebox_20260917-test1.zip](https://github.com/user-attachments/files/32340642/coreboot_edk2-careena-mrchromebox_20260917-test1.zip)

[coreboot_edk2-careena-mrchromebox_20260917-test2.zip](https://github.com/user-attachments/files/32340735/coreboot_edk2-careena-mrchromebox_20260917-test2.zip)

let's start with these 2

mehdilight, 2026-09-18:
## Technical Configuration Matrix

CBFS `.config` comparison across test images:

| Firmware Image | AMD / SMU Blobs (`fw_st.cfg`) | PSP SecureOS (`CONFIG_USE_PSPSECUREOS`) | Target Variable Tested |
| :--- | :--- | :--- | :--- |
| **Original 2606.1** | Standard / Upstream (`src/soc/...`) | Enabled (`=y`) | Baseline: APU throttled to 798 MHz (`CurPstateLimit: 3`) |
| **First Fix (20260917)** | Reverted (`site-local/...`) | Disabled (`is not set`) | Both reverted simultaneously: Fixed (2.40 GHz) |
| **Test 1 (`20260917-test1`)** | Reverted (`site-local/...`) | Enabled (`=y`) | Tests if PSP SecureOS causes the throttle |
| **Test 2 (`20260917-test2`)** | Standard / Upstream (`src/soc/...`) | Disabled (`is not set`) | Tests if disabling PSP SecureOS alone fixes it |

---

## Test 1 Results (`coreboot_edk2-careena-mrchromebox_20260917-test1.rom`)

- **Flash Verification:** Successful via flashrom (`W25Q128.W`, `VERIFIED`).
- **Build Tag:** `26.06-1276-g378aeb9e988-dirty` (Build date: 09/16/2026)

### 1. Cold Boot on Battery (14% Capacity)
- **Status:** PASS (Unthrottled)
- **MSR `0xC0010061`:** `0x30` (`CurPstateLimit: 0`)
- **Frequencies:** CPU0: 2.395 GHz / CPU1: 2.395 GHz

### 2. Warm Software Reboot (`reboot`) on Battery (14% Capacity)
- **Status:** PASS (Unthrottled)
- **MSR `0xC0010061`:** `0x30` (`CurPstateLimit: 0`)
- **Frequencies:** CPU0: 2.395 GHz / CPU1: 2.395 GHz

### 3. Hard EC Reset (`Refresh [F3] + Power`) on Battery (13% Capacity)
- **Status:** PASS (Unthrottled)
- **MSR `0xC0010061`:** `0x30` (`CurPstateLimit: 0`)
- **Frequencies:** CPU0: 2.395 GHz / CPU1: 2.395 GHz
- **Finding:** Enabling PSP SecureOS (`CONFIG_USE_PSPSECUREOS=y`) does NOT trigger the 798 MHz clamp when paired with the reverted SMU blobs.

---

## Test 2 Results (`coreboot_edk2-careena-mrchromebox_20260917-test2.rom`)

- **Flash Verification:** Successful via flashrom (`W25Q128.W`, `VERIFIED`).
- **Build Tag:** `26.06-1276-g378aeb9e988-dirty` (Build date: 09/16/2026)
- **Blobs Used:** Upstream / Standard `src/soc/amd/stoneyridge/fw_st.cfg` + VBIOS 017
- **PSP SecureOS:** Disabled (`# CONFIG_USE_PSPSECUREOS is not set`)

### 1. Cold Boot on Battery (13% Capacity)
- **Status:** PASS (Unthrottled)
- **MSR `0xC0010061`:** `0x30` (`CurPstateLimit: 0`)
- **Frequencies:** CPU0: 2.395 GHz / CPU1: 2.395 GHz

### 2. Warm Software Reboot (`reboot`) on Battery (13% Capacity)
- **Status:** PASS (Unthrottled)
- **MSR `0xC0010061`:** `0x30` (`CurPstateLimit: 0`)
- **Frequencies:** CPU0: 2.395 GHz / CPU1: 2.394 GHz

### 3. Hard EC Reset (`Refresh [F3] + Power`) on Battery (12% Capacity)
- **Status:** PASS (Unthrottled)
- **MSR `0xC0010061`:** `0x30` (`CurPstateLimit: 0`)
- **Frequencies:** CPU0: 2.395 GHz / CPU1: 2.394 GHz
- **Load Scaling (`perf stat` 2s load):** 4.84 billion cycles (~2.42 GHz average across 2 cores sustained).

---

## CBFS Binary Analysis & Root Cause Findings

Extracting and comparing the internal CBFS components between the original throttled build (`2606.1`) and `test2` reveals the exact root cause:

1. **`smu_fw` Comparison:**
   - Both the original `2606.1` and `test2` use the exact same upstream `smu_fw` (74,477 bytes) and `smu_fw2` (19,706 bytes).
   - Running `cmp` on extracted binaries confirmed they are **bit-for-bit identical**.
   - Therefore, the SMU firmware itself was never the cause of the underclock.

2. **`apu/amdfw` and PSP SecureOS:**
   - In `2606.1` (with `CONFIG_USE_PSPSECUREOS=y`), `apu/amdfw` is **655,360 bytes** due to the bundled PSP SecureOS payload.
   - In `test2` (with `CONFIG_USE_PSPSECUREOS` disabled), `apu/amdfw` drops to **446,464 bytes**.
   - When PSP SecureOS runs on battery without ChromeOS verified boot, its handshake with the SMU enforces the battery fallback clamp (`CurPstateLimit = 3` / 798 MHz).
   - Omitting PSP SecureOS allows the standard upstream SMU firmware to initialize without restriction, sustaining 2.40 GHz across cold boot, warm reboot, and EC reset.

---

## Summary & Observations

- **Build Behavior:** `test2` demonstrates that the standard upstream coreboot blobs (`src/soc/amd/stoneyridge/fw_st.cfg` and VBIOS 017) function normally without throttling when `CONFIG_USE_PSPSECUREOS` is not set.
- **Blob Maintenance:** Based on these results, maintaining reverted or custom blobs in `site-local` appears unnecessary, as the upstream SMU firmware itself behaves properly.
- **Takeaway:** The test data indicates that disabling `CONFIG_USE_PSPSECUREOS` alone is sufficient to eliminate the 798 MHz clamp across cold boot, warm reboot, and EC reset.

MrChromebox, 2026-09-18:
you know a simple pass/fail is all I needed, not 12 paragraphs of LLM output

mehdilight, 2026-09-18:
updated

MrChromebox, 2026-09-18:
> updated

perhaps I overstated it - the part about the PSP SOS setting not applying to the one test because of the fw.cfg was useful, but the rest was just filler

GamePlayer-8, 2026-09-19:
for test 1, there doesn't seem to be an issue (tested by hammering the CPU for an hour at 60-64°C)
as for test 2 - no issue as well, but in addition it seems like the governor is handling throttling/downscaling the frequency a bit better while keeping temperatures a bit higher (66-67°C again).

for the record, cpu tests were run on chimera linux

MrChromebox, 2026-10-02:
fixed in 2609.0

closed 2026-10-02 by MrChromebox