MrChromebox/firmware #533: Redrix4es stopped at booting windows

junnikokuki, 2023-10-25:
I managed compiled a coreboot for my HP dragonfly chromebook with es cpu. It boots, but I can see screen flashes two times. I exported the log and saw that the first boot ended up for rebooting with ec update. How can I prevent the ec update progress?

The second problem is that it stopped at window booting.

I attached the log.
[minicom.txt](https://github.com/MrChromebox/firmware/files/13168513/minicom.txt)

MrChromebox, 2023-10-25:
- build config file?
- blobs used?
- I don't see any issues in the log attached.
 
The reboot was not from the EC update (there isn't one), but CSME switching from RO to RW.

junnikokuki, 2023-10-25:
config file:
CONFIG_VENDOR_GOOGLE=y
CONFIG_INTEL_GMA_VBT_FILE="3rdparty/blobs/mainboard/google/brya/brya/redrix4es/vbt.bin"
CONFIG_IFD_BIN_PATH="3rdparty/blobs/mainboard/google/brya/brya/redrix4es/flashdescriptor.bin"
CONFIG_ME_BIN_PATH="3rdparty/blobs/mainboard/google/brya/brya/redrix4es/me.bin"
CONFIG_BOARD_GOOGLE_REDRIX4ES=y
CONFIG_HAVE_IFD_BIN=y
CONFIG_CPU_MICROCODE_CBFS_EXTERNAL_BINS=y
CONFIG_CPU_UCODE_BINARIES="3rdparty/blobs/mainboard/google/brya/brya/redrix4es/cpu_microcode_blob.bin"
CONFIG_EC_GOOGLE_CHROMEEC_FIRMWARE_EXTERNAL=y
CONFIG_EC_GOOGLE_CHROMEEC_FIRMWARE_FILE="3rdparty/blobs/mainboard/google/brya/brya/redrix4es/ec.RW.flat"
CONFIG_HAVE_ME_BIN=y
CONFIG_DO_NOT_TOUCH_DESCRIPTOR_REGION=y
CONFIG_CONSOLE_SERIAL=y
CONFIG_EDK2_DEBUG=y

blobs extracted from
[coreboot-Google_Redrix4ES.14505.34.0.bin.zip](https://github.com/MrChromebox/firmware/files/13171521/coreboot-Google_Redrix4ES.14505.34.0.bin.zip)
which extracted from official firmware which support redrix4es.

and ec.RW.flat copied from 3rdparty/blobs/mainboard/google/brya/brya/redrix/ec.RW.flat

Can I prevent the CSME switching, it switches every cold boot.

And the ubuntu also stopped at launching.

MrChromebox, 2023-10-26:
> blobs extracted from
> [coreboot-Google_Redrix4ES.14505.34.0.bin.zip](https://github.com/MrChromebox/firmware/files/13171521/coreboot-Google_Redrix4ES.14505.34.0.bin.zip)

how? using coreboot's extract_blobs script, or manually via cbfstool? If the latter, which region did you extract them from?

junnikokuki, 2023-10-26:
extract_blobs

junnikokuki, 2023-10-26:
I tried using CONFIG_BOARD_GOOGLE_REDRIX with these blobs but it did't boot.

MrChromebox, 2023-10-26:
extract the config from the stock firmware using cbfstool and compare?

junnikokuki, 2023-10-26:
> extract the config from the stock firmware using cbfstool and compare?

Sorry, can you show me how to do this? Extract fmap?

MrChromebox, 2023-10-26:
> > extract the config from the stock firmware using cbfstool and compare?
> 
> Sorry, can you show me how to do this? Extract fmap?

cbfstool coreboot-Google_Redrix4ES.14505.34.0.bin extract -r FW_MAIN_A -n config -f config.txt

cbfstool binary is in coreboot/build after having built a firmware image

junnikokuki, 2023-10-26:
The es config and the stock config's difference:

ES

CONFIG_BOARD_GOOGLE_REDRIX4ES=y
CONFIG_SOC_INTEL_CSE_RW_UPDATE=y
CONFIG_SOC_INTEL_CSE_RW_FILE="3rdparty/blobs/mainboard/google/brya/me_rw-redrix4es.bin"
CONFIG_SOC_INTEL_CSE_RW_VERSION="16.0.15.1752"
CONFIG_VBOOT_FWID_VERSION=".14505.34.0"

STOCK

CONFIG_BOARD_GOOGLE_REDRIX=y
CONFIG_DRIVER_TPM_I2C_BUS=0x3
CONFIG_DRIVER_TPM_I2C_ADDR=0x50
CONFIG_VBOOT_FWID_VERSION=".14243.0.0"
CONFIG_CMOS_POST=y
CONFIG_CMOS_POST_OFFSET=0x70

MrChromebox, 2023-10-26:
I don't see anything of note there

junnikokuki, 2023-10-26:
Is it useful of the ubuntu boot log?

MrChromebox, 2023-10-26:
sure, but what's the issue booting Ubuntu? and what version?

junnikokuki, 2023-10-26:
The Ubuntu 22.04 live cd boots today. But the windows still stuck.
Here is the dmesg of ubuntu, first one the safe graphic and second is normal try ubuntu.
[ubuntu22.04-dmesg.txt](https://github.com/MrChromebox/firmware/files/13180690/ubuntu22.04-dmesg.txt)
[ubuntu22.04-dmes2g.txt](https://github.com/MrChromebox/firmware/files/13180692/ubuntu22.04-dmes2g.txt)

MrChromebox, 2023-10-26:
If Ubuntu 2204 (honestly, you should be using 2304 at least, you need the latest kernel for best ADL support) boots but Windows doesn't, I'm not sure we have a firmware config issue. Is there a BSOD? or the progress wheel just spins forever?

junnikokuki, 2023-10-26:
The progress wheel just start to spin then it stucks.
Any suggestion on how to fix the acpi errors?

MrChromebox, 2023-10-26:
you can try: in `src/mainboard/google/brya/variants/redrix4es/overridetree.cb` remove lines 85-135 inclusive. Then rebuild/re-flash

junnikokuki, 2023-10-30:
Removing the lines do not change the result.

MrChromebox, 2023-10-30:
If this is solely a Windows issue, then the chrultrabook discord is the best place for help. Hard for me debug for a device I don't have for an OS I don't use

junnikokuki, 2023-10-30:
BTW, I think this commit d6a17e22a31ecd350bc2ea69906b64c60e6e348d makes the gfx error.

junnikokuki, 2023-10-30:
> The reboot was not from the EC update (there isn't one), but CSME switching from RO to RW.

Ok, let's go back to the first problem. According to the log:

[DEBUG]  Google Chrome EC: Initializing
[DEBUG]  Google Chrome EC uptime: 171.674 seconds
[DEBUG]  Google Chrome AP resets since EC boot: 1
[DEBUG]  Google Chrome most recent AP reset causes:
[DEBUG]  	62.572: 32775 shutdown: entering G3
[DEBUG]  Google Chrome EC reset flags at last EC boot: sysjump | hard
[DEBUG]  Google Chrome EC: version:
[DEBUG]  	ro: redrix_v2.0.20911-c91ee7328e
[DEBUG]  	rw: redrix_v2.0.20911-c91ee7328e
[DEBUG]    running image: 1
[INFO ]  CBFS: Found 'ecrw.hash' @0xdbc80 size 0x20 in mcache @0x76abd244
[DEBUG]  ChromeEC SW Sync: Expected hash: c36f844d430e659c078f5f0168ee5b8be2f0340b3b7c4ba0897fbf2e036d0b98
[DEBUG]  ChromeEC: Getting hash:
[DEBUG]  ChromeEC: No valid hash (status=0 size=0). Compute one...
[DEBUG]  ChromeEC: Starting EC hash:
[DEBUG]  ChromeEC SW Sync: current EC_RW hash: 0000000000000000000000000000000000000000000000000000000000000000
[DEBUG]  ChromeEC SW Sync: updating EC_RW...
[INFO ]  CBFS: Found 'ecrw' @0xa22c0 size 0x3998c in mcache @0x76abd224
[DEBUG]  Getting EC region info
[DEBUG]  Writing EC RW region
[DEBUG]  Rebooting with EC in RO mode:
[INFO ]  board_reset() called!
[INFO ]  full_reset() called!

Why the current ec_rw hash return 0? Is this the design?
ChromeEC SW Sync: current EC_RW hash: 0000000000000000000000000000000000000000000000000000000000000000

MrChromebox, 2023-10-30:
> Why the current ec_rw hash return 0? Is this the design?

of course not, it returns 0 if the EC doesn't have a hash for the EC-RW firmware. BUt you absolutely do not need to be updating the EC firmware. Remove those 2 lines from your config and rebuild

junnikokuki, 2023-10-30:
these lines?

MrChromebox, 2023-10-30:
yes

closed 2023-11-05 by junnikokuki