MrChromebox/firmware #534: Lenovo C13 Yoga Chromebook Wifi(AX200)  can not resume from suspend

emf37, 2023-10-27:
hello MrChromebox,
I have a C13 Yoga chromeBook and I have flashed 2023-10-13 full UEFI rom by Firmware Utility Script. When I switch to UEFI, I installed fedora 38 for laptop, and i found when I wake up laptop from suspend(S3) the WiFi was unavailable. dmesg showed "iwlwifi 0000:01:00.0: Unable to change power state from D3hot to D0, device inaccessible" and trace. I have tried disabling the iwlwifi power save mode, reinstalling another distro (openSUSE Tumbleweed, Ubuntu, AOSC) and windows 10, they both show this bug.  A community people told me this bug might be a firmware bug, so I decided to report it here. I have attached a dmesg log file from fedora, if you need more  info, please let me know.

regards
[dmesg_C13_yoga_20231027.log](https://github.com/MrChromebox/firmware/files/13187521/dmesg_C13_yoga_20231027.log)

dawenhaozx, 2023-10-27:
Bro, glad we meet again. I have the same situation as you. It happened on yoga 13c 3500c in windows 11 (full uefi rom 2023-10-12).  I replaced the original rtk 8822ce with intel 3168ac and tried many versions of WiFi drivers.  After waking up from sleep, the result is that WiFi and Bluetooth disappear in the device management.  Both WiFi and Bluetooth can only work after restarting windows 11.  By the way, I would like to thank the author and other developers for their hard work. Apart from this bug, everything else is working great.

emf37, 2023-10-27:
> Bro, glad we meet again. I have the same situation as you. It happened on yoga 13c 3500c in windows 11 (full uefi rom 2023-10-12). I replaced the original rtk 8822ce with intel 3168ac and tried many [...]

thanks bro, my SKU is C13 Yoga 3700c. I'm just finding an Intel 7260AC card and would like to replace the ax200. Thanks for your comment so I don't have to test another intel wireless card. Could I ask your rtk card have worked well？btw, I'm really appreciating the author too, thanks for these great works

MrChromebox, 2023-10-27:
known issue, will be fixed in upcoming 4.21.1 release (hopefully this weekend - in testing now)

emf37, 2023-10-27:
> known issue, will be fixed in upcoming 4.21.1 release (hopefully this weekend - in testing now)

that's would be nice!  appreciation for your work!

closed 2023-10-27 by emf37

dawenhaozx, 2023-10-27:
> thanks bro, my SKU is C13 Yoga 3700c. I'm just finding an Intel 7260AC card and would like to replace the ax200. Thanks for your comment so I don't have to test another intel wireless card. Could I [...]

Yoga c13 has rtk 8822ce(wifi5)/intel ax200(wifi6).  The rtk8822ce is a cheaper option and is quite average.  Your ax200 is better.And yoga c13 should only be compatible with these two wifi cards after restoring it to chromeos.  For yoga c13, intel 3168 ac cannot turn on wifi in chromeos.