MrChromebox/firmware #536: ThinkPad C13 Yoga Chromebook the trackpoint and mouse buttons sometimes become unresponsive, chrome ec driver conflict

dawenhaozx, 2023-10-30:
Hello, MrChromebox. First of all, I want to express my gratitude for your efforts and those of other developers. 
Model: Yoga C13 3500c
UEFI: morphius-mrchromebox_20231029 / UEFI Full ROM firmware
OS: Windows 11
The specific issue I'm experiencing is as follows: the touchscreen, keyboard, the power button, and volume keys all work normally, trackpoint and mouse buttons (which may become unresponsive).

After installing Coolstar's Chrome EC drivers for key mapping, when key mapping is functioning correctly.
The volume keys become unresponsive, trackpoint and mouse buttons may become unresponsive.
After a particular reboot, only the touchscreen remains functional, and the keyboard briefly works after waking from sleep (for about 1-2 seconds). Finally, it's only after completely uninstalling the Chrome EC drivers that I can restore normal functionality, but the trackpoint and mouse buttons (which may become unresponsive).

dawenhaozx, 2023-10-30:
> this is pretty clearly an OS/driver issue, not a firmware one, so not something I can help with. I'm not sure which of coolstar's repos would be best to open an issue in, so I'd recommend reaching out on the chrultrabook Discord for help there.

Sorry, my description was not clear.  Update MrChromebox-4.21.1 Release
 (2023.10.29), I reinstalled Windows 11. When the driver was not installed, the trackpoint (red dot) and the buttons below corresponding to the left, middle and right mouse buttons failed.  If they don't work after turning on the computer, they may become normal after restarting.  But it does not appear like this in ubuntu. But in the previous version
 MrChromebox-4.21.0 Release
 (2023.10.01) Except for the bug that the wifi fails after waking up, everything about the keyboard works

MrChromebox, 2023-10-30:
> But it does not appear like this in ubuntu.

I don't understand what you are saying here. Do the trackpoint and mouse buttons work under Ubuntu? All the time?

junnikokuki, 2023-10-30:
I think he is right. I previously used the 4k touchscreen fixed version you compiled for me on 12 oct, the trackpoint works. Today I updated the coreboot 1029, when the c13 goes to sleep, the power light will not breath as previous version. And when I resume from sleep, the trackpoint and three keys along with touch screen will not response. I think sth changed after the 4k touchscreen fix commit. It is not a driver related issus but bios.

MrChromebox, 2023-10-30:
yes, there were some changes affecting the keyboard, the previous workaround was what killed WiFi on resume from sleep, so we had to implement a different fix. Hopefully no KB init workaround will be necessary for 4.22, but waiting on a fixed signed PSP component from Google.

dawenhaozx, 2023-10-31:
> > But it does not appear like this in ubuntu.
> 
> I don't understand what you are saying here. Do the trackpoint and mouse buttons work under Ubuntu? All the time?

Yes, I booted Ubuntu through a USB flash drive to flash the new firmware, and the trackpoint and three keys worked. But when running the Windows 11 installation disk, they may fail. Thank you and the brother above, thank you for your help, I hope this can be solved in the future.

junnikokuki, 2023-10-31:
> Yes, I booted Ubuntu through a USB flash drive to flash the new firmware, and the trackpoint and three keys worked. But when running the Windows 11 installation disk, they may fail. Thank you and [...]

I suggest you flash the 1012 version and turn off sleep if you want to use windows

dawenhaozx, 2023-10-31:
> > > > But it does not appear like this in ubuntu.
> > > 
> > > 
> > > I don't understand what you are saying here. Do the trackpoint and mouse buttons work under Ubuntu? All the time?
> > 
> > 
> > Yes, I booted Ubuntu through a USB flash drive to flash the new firmware, and the trackpoint and three keys worked. But when running the Windows 11 installation disk, they may fail. Thank you and the brother above, thank you for your help, I hope this can be solved in the future.
> 
> I suggest you flash the 0930 version and turn off sleep if you want to use windows

Thanks for reminding me, this is a great idea!!!