MrChromebox/firmware #540: Battery will not charge after flashing C302C

tquid, 2023-11-04:
Hi, I've been able to flash and successfully installed a couple different OSes, but the power indicator flashes alternately amber and yellow, and the battery never takes a charge. I installed the complete UEFI coreboot, since this model is end-of-lifed.

I have tried some reset Vulcan nerve pinches such as holding "reload" (F3) while powering on, or holding the power button and volume-up, to no avail. I have also disconnected and reconnected the battery.

Assuming the ROM over-write may have made the usual key/button combos in accessible, is there another way to get that reset to happen?

tquid, 2023-11-09:
Hi, and thanks for patiently waiting,

I fear you may be right about the battery connection, or it may just be the battery has died. I have to decide if it's worth it to try a new batter at $45.

Just in case, though, I did extract the requested log, attached herewith. Please let me know if anything stands out.
[cbmem.log](https://github.com/MrChromebox/firmware/files/13312992/cbmem.log)

MrChromebox, 2023-11-09:
the cbmem log shows your device running the correct ECRW firmware, so I'd wager it is an issue with the battery or connector

tquid, 2023-11-11:
New battery it is, then! Thanks very much for helping out so quickly and thoroughly.

closed 2023-11-11 by tquid