MrChromebox/firmware #543: Tablet mode cannot be used on DELL 7410 (drallion)

geochensx, 2023-11-08:
1.Tablet mode cannot be used on DELL 7410 (drallion)
2.The laptop battery is still 20% to 30%, and the laptop will suddenly power off and shut down. I don't know if it's my computer battery or if everyone is in the same situation.

Mercu7io, 2024-02-12:
Same issue on my drallion with the firmware 4-22.2

igyurisa, 2024-06-14:
It's exactly the same problem I encountered. Everything else about this notebook is perfect, except for these two problems that need to be solved.

igyurisa, 2024-06-14:
@MrChromebox I'd love your help in fixing these two issues, thank you very much

MrChromebox, 2024-06-14:
I don't have this device to test, and nothing jumps out in the code/commit history. but nobody here has even bothered to report what OS they are using so...

igyurisa, 2024-06-15:
> I don't have this device to test, and nothing jumps out in the code/commit history. but nobody here has even bothered to report what OS they are using so...我没有这个设备来测试，并且代码/提交历史记录中没有任何内容跳出。但这里没有人愿意报告他们正在使用什么操作系统，所以......

This problem occurs both under win10 and win11

MrChromebox, 2024-06-15:
> This problem occurs both under win10 and win11

the chrultrabook forums are the place to go for Windows support

igyurisa, 2024-06-15:
> > This problem occurs both under win10 and win11
> 
> the chrultrabook forums are the place to go for Windows support

https://forum.chrultrabook.com/t/battery-error-on-dell-latitude-7410-chrome/413

Mercu7io, 2024-06-15:
Both are not fixed. and Since Coolstar don't have a drallion anymore, it might be not fixed

geochensx, 2024-06-19:
I also tested fydeos, and it didn't work either.

Mercu7io, 2024-06-20:
It's getting worse on my drallion. Now, it start at 46%
I installed MacOS, and I have also an alert battery is not working well

closed 2026-11-04 by MrChromebox