MrChromebox/firmware #544: Lenovo Yoga Chromebook C630 / Pantheon (Intel Kabylake) - USB get off

InformatiqueCIASalOuestDeRennes, 2023-11-08:
We have 5x Lenovo Yoga Chromebook C630 / Pantheon (Intel Kabylake).

After flashing full-firmware, internal USB (type A) port is working. I install the OS (Zorin OS or Deban Bookworm), everything is OK.
I power off the Chromebook, and after some hours, I power on the Chromebook but USB (type A) port is not working.
USB-C is working : if I connect an adapter to port on left or right, no problem. I can plug an USB-Key on adapter and use the peripheral.
If I connect the USB-Key on USB-A, this does not work.
To get it working again, I have to poweroff the Chromebook, then press ESC+REFRESH+POWER then USB (type A) port is working.
Firmware issue ?

UNTUNGKSR, 2024-01-24:
Hello, I've got the same issue here. My Chromebook is known as C340-15 on the case, and when I installed the mrchromebox BIOS, it said my device was C340 Laser Motherboard, but after installing Windows, and checking with the CPUID, it said my device was C630 Yoga with Pantheon Mobo. While installing the driver I had to mix and match these 2 types of driver to get all of the IO devices to work, except for the USB A and the sound (paid driver).
And then I have another question here, can I install Windows on drive 1 (126 SD card) rather than install it on drive 0 (32Gb Emmc), Because with 32gb Emmc I can't add any applications. Thx b4

MrChromebox, 2024-01-24:
> and when I installed the mrchromebox BIOS, it said my device was C340 Laser Motherboard, but after installing Windows, and checking with the CPUID, it said my device was C630 Yoga with Pantheon Mobo

this is not possible, they are completely different platforms, and the firmware for one would not boot on the other. Whatever the firmware says in the settings menu is correct

UNTUNGKSR, 2024-01-24:
Okay. Maybe it was my mistake. So Sir, can you help me with my USB A port issue? Is there any tips to get my USB A port function?

Thx again.

closed 2026-11-04 by MrChromebox