MrChromebox/firmware #545: Full ROM bricks Thinkpad C13 Chromebook

FriederHannenheim, 2023-11-12:
As already mentioned in https://github.com/MrChromebox/scripts/issues/326 the Lenovo Thinkpad C13 Chromebook fails to POST after a full ROM flash. After I flashed and restarted the device the keyboard light and fan turned on but nothing else happens. It has been like this for 15 minutes now so I'm assuming it's bricked. 

This is actually good for me since now I have an excuse to buy another Laptop so I won't have to use this piece of shit anymore.

MrChromebox, 2023-11-12:
ok?

FriederHannenheim, 2023-11-12:
I'm just saying maybe a Full ROM flash should not be recommended or enabled for that device since it bricks it

MrChromebox, 2023-11-12:
my C13 is running it just fine

FriederHannenheim, 2023-11-14:
Maybe it's just the MMC models then

XperfectTR, 2023-11-24:
I had the same issue, and I ended up dismantling the device. I rewrote the BIOS from the backup BIOS file, and now it's working fine. The only way to use Windows on this computer is through RWL, but that's only possible with an external SSD. The touchscreen and touchpad aren't working due to ACPI issues.

MrChromebox, 2023-11-24:
> The only way to use Windows on this computer is through RWL, but that's only possible with an external SSD. The touchscreen and touchpad aren't working due to ACPI issues.

that's not a problem I have any intention of solving. Windows via RWL is a coolstar thing

MrChromebox, 2023-11-27:
new firmware pushed for all Picasso boards which should resolve the issue

closed 2023-11-27 by MrChromebox