MrChromebox/firmware #546: Unable to shutdown, turns ON again until second shutdown - Lenovo Flex 5i (LILLIPUP)

marcosilva85, 2023-11-13:
Shutdown is not working properly because it just turns ON the laptop again.
I see systemctl properly logging the execution of shutdown procedures, the laptop screen and status LED even turns off, but after 1 second or two it turns ON by itself.

When I execute a shutdown the second time, the laptop then stays OFF as expected.
I only have the power supply connected. No ETHERNET or wake or lan or anything...

Latest MrChromebox-4.21.1, 6.5.9 and also 6.6 Linux kernel.
Did multiple trials, it does it on any DE (KDE/gnome,GDM, systemd shutdowns...).

fappifap, 2024-03-27:
i have the same problem with my Lenovo IdeaPad 5 (Lindar)

jmandawg, 2024-05-19:
I have the same issue on my Lenovo flex 5.  It even happens during hibernate.  It will hibernate then turn right back on

closed 2026-11-04 by MrChromebox