MrChromebox/firmware #550: Touchscreen not working:  Dell Latitude 7410 2-in-1 Chromebook

rustymiller, 2023-11-26:
First of all, thanks @MrChromebox for all the information on how to get another OS on a Chromebook - really appreciate the work you put into this!

**Machine**:  Dell Latitude 7410 2-in-1 Chromebook
**Motherboard**:  Drallion
**Boot Mode**:  RW_LEGACY (Ctrl+L on the Developer Mode white screen) and booting with TianoCore  (I used the "ChromeOS Firmware Utility Script" to install it.)
**TouchScreen Device**:  WCOM48E2, Wacom Touchscreen
**OS**: Tried both Ubuntu 22.04.3 and Mint 21.2 (x64) booted from RW_LEGACY

Side note: when using "ChromeOS Firmware Utility Script" the option for RW_LEGACY update was greyed out, but I was still able to select it:

[image]

**Result**: The touchscreen doesn't work.

dmesg logs:
```
...
[    1.178420] cros_ec_lpcs cros_ec_lpcs.0: couldn't reserve memmap region
[    1.178440] cros_ec_lpcs: probe of cros_ec_lpcs.0 failed with error -16
...
[    2.226573] i2c_designware i2c_designware.0: controller timed out
[    3.250593] i2c_designware i2c_designware.0: controller timed out
[    4.274587] i2c_designware i2c_designware.0: controller timed out
[    5.298597] i2c_designware i2c_designware.0: controller timed out
[    5.298693] i2c_hid_acpi i2c-WCOM48E2:00: failed to reset device.
[    5.298732] i2c_hid_acpi i2c-WCOM48E2:00: failed to reset device: -110
[    6.322697] i2c_designware i2c_designware.0: controller timed out
[    7.346613] i2c_designware i2c_designware.0: controller timed out
[    7.346710] i2c_hid_acpi i2c-WCOM48E2:00: failed to change power setting.
[    8.370655] i2c_designware i2c_designware.0: controller timed out
[    9.394610] i2c_designware i2c_designware.0: controller timed out
[    9.394721] i2c_hid_acpi i2c-WCOM48E2:00: failed to change power setting.
[   10.418613] i2c_designware i2c_designware.0: controller timed out
[   11.442643] i2c_designware i2c_designware.0: controller timed out
[   11.442741] i2c_hid_acpi i2c-WCOM48E2:00: failed to change power setting.
[   12.466635] i2c_designware i2c_designware.0: controller timed out
[   13.490636] i2c_designware i2c_designware.0: controller timed out
[   13.490733] i2c_hid_acpi i2c-WCOM48E2:00: failed to change power setting.
[   14.514705] i2c_designware i2c_designware.0: controller timed out
[   14.514706] i2c_hid_acpi i2c-WCOM48E2:00: can't add hid device: -110
[   14.531220] i2c_hid_acpi: probe of i2c-WCOM48E2:00 failed with error -110
[   14.541007] ish-loader {C804D06A-55BD-4EA7-ADED-1E31228C76DC}: Direct firmware load for intel/drallion_ish.bin failed with error -2
[   14.541013] ish-loader {C804D06A-55BD-4EA7-ADED-1E31228C76DC}: ISH host firmware load failed -2
...
[   15.832258] dell_smbios: Unable to run on non-Dell system
...
[   16.434552] i2c_designware i2c_designware.0: controller timed out
[   16.434566] mip4_ts i2c-MLFS0000:00: nothing at this address
...
```

When booting into ChromeOS where the touchscreen is working, the touch screen device info is:

```
		device pci 15.0 on
			chip drivers/i2c/hid
				register "generic.hid" = ""WCOM48E2""
				register "generic.desc" = ""Wacom Touchscreen""
				register "generic.irq" = "ACPI_IRQ_LEVEL_LOW(GPP_C23_IRQ)"
				register "generic.probed" = "1"
				register "generic.stop_gpio" = "ACPI_GPIO_OUTPUT_ACTIVE_LOW(GPP_E7)"
				register "generic.stop_delay_ms" = "20"
				register "generic.enable_gpio" = "ACPI_GPIO_OUTPUT_ACTIVE_HIGH(GPP_B21)"
				register "generic.enable_delay_ms" = "55"
				register "generic.has_power_resource" = "1"
				register "generic.disable_gpio_export_in_crs" = "1"
				register "hid_desc_reg_offset" = "0x1"
				register "generic.device_present_gpio" = "GPP_B4"
				register "generic.device_present_gpio_invert" = "1"
				device i2c 0A on end
			end
			chip drivers/i2c/hid
				register "generic.hid" = ""ELAN900C""
				register "generic.desc" = ""ELAN Touchscreen""
				register "generic.irq" = "ACPI_IRQ_EDGE_LOW(GPP_C23_IRQ)"
				register "generic.probed" = "1"
				register "generic.stop_gpio" =
							"ACPI_GPIO_OUTPUT_ACTIVE_LOW(GPP_E7)"
				register "generic.stop_delay_ms" = "10"
				register "generic.enable_gpio" =
							"ACPI_GPIO_OUTPUT_ACTIVE_HIGH(GPP_B21)"
				register "generic.enable_delay_ms" = "55"
				register "generic.has_power_resource" = "1"
				register "hid_desc_reg_offset" = "0x01"
				register "generic.device_present_gpio" = "GPP_B4"
				register "generic.device_present_gpio_invert" = "1"
				device i2c 10 on end
			end
			chip drivers/i2c/generic
				register "hid" = ""MLFS0000""
				register "desc" = ""Melfas Touchscreen""
				register "irq" = "ACPI_IRQ_EDGE_LOW(GPP_C23_IRQ)"
				register "probed" = "1"
				register "stop_gpio" = "ACPI_GPIO_OUTPUT_ACTIVE_LOW(GPP_E7)"
				register "stop_delay_ms" = "10"
				register "enable_gpio" = "ACPI_GPIO_OUTPUT_ACTIVE_HIGH(GPP_B21)"
				register "enable_delay_ms" = "55"
				register "has_power_resource" = "1"
				register "device_present_gpio" = "GPP_B4"
				register "device_present_gpio_invert" = "1"
				device i2c 34 on end
			end
		end # I2C #0
```

Based on another thread, I tied unloading and reloading **i2c_hid**, **i2c_hid_acpi** and **mip4_ts**, but that didn't help.

I didn't try to "Full ROM" update (wanted to avoid the risk of bricking the device).  Is it expected that the Full ROM update would address a problem like this?

MrChromebox, 2023-11-26:
there's nothing that can be done with RW_LEGACY really (because it's just a bootloader, and you're still using the stock firmware), unless you want to use something like OpenCore to modify the ACPI tables before booting Linux (which is out of my wheelhouse). Everything works properly on the UEFI Full ROM AFAIK; coolstar has (had?) one and assisted with fixing everything

rustymiller, 2023-11-26:
Thanks!  I did the "Full ROM" update and the the touchscreen is now working.  These are the only errors on boot now (in case this is useful to someone):

```
[    0.148516] x86/cpu: SGX disabled by BIOS.
[    0.865190] cros_ec_lpcs: unsupported system.
[    1.494457] ish-loader {C804D06A-55BD-4EA7-ADED-1E31228C76DC}: ISH host firmware load failed -2
[    2.248979] cros_ec_lpcs: unsupported system.
[    4.385331] Bluetooth: hci0: Malformed MSFT vendor event: 0x02
```

At this point, the only problem I notice is that the when the screen is folded over in table mode, the keyboard and touchpad are still active (and thus getting unwanted input while holding it).

BTW, there is a bug logged on the first boot after rebooting from the "Full ROM" update in case it's useful (full trace: [stacktrace.txt](https://github.com/MrChromebox/firmware/files/13468034/stacktrace.txt) ):

```
Nov 26 19:35:47 chr-Drallion kernel: [Firmware Bug]: Page fault caused by firmware at PA: 0xffffffff0c9375e0
Nov 26 19:35:47 chr-Drallion kernel: WARNING: CPU: 0 PID: 9 at arch/x86/platform/efi/quirks.c:735 efi_crash_gracefully_on_page_fault+0x63/0x110
Nov 26 19:35:47 chr-Drallion kernel: Modules linked in: snd_seq_midi snd_seq_midi_event snd_rawmidi i915(+) coretemp btusb drm_buddy iwlmvm(+) snd_seq btrtl kvm_intel ttm btbcm snd_seq_device uvcvideo >
Nov 26 19:35:47 chr-Drallion kernel:  int340x_thermal_zone int3400_thermal acpi_thermal_rel mac_hid sch_fq_codel msr parport_pc ppdev lp drm parport efi_pstore ip_tables x_tables autofs4 intel_ishtp_lo>
Nov 26 19:35:47 chr-Drallion kernel: CPU: 0 PID: 9 Comm: kworker/u16:0 Not tainted 6.2.0-37-generic #38~22.04.1-Ubuntu
Nov 26 19:35:47 chr-Drallion kernel: Hardware name: Google Drallion/Drallion, BIOS MrChromebox-4.21.1 10/29/2023
```