MrChromebox/firmware #552: Low VRAM

Striker770, 2023-11-28:
Hello, 

Having Lenovo ThinkPad C13 Yoga Chromebook and installed the Mr.Chromebox firmware (Full UEFI). When i tried to install Windows, i saw that the VRAM is just 32MB. Verified it from AMD Control Panel. Is there any way to increase the VRAM?

On my HP 3250U, it has 8GB RAM but the VRAM is 2048MB.

Thanks

ys202, 2024-03-17:
> closing this as all PCO boards will have fixed 512MB VRAM going forward

Hello,
Thanks for your great work.

But I still have only 32Mb VRAM on freshly installed WIN10 with the actual Mr.Chromebox firmware (Full UEFI rom from your website) on Lenovo ThinkPad C13 Yoga Chromebook (Ryzen 3500c, 16Gb RAM).  

I'm trying to find a proper config or option to increase VRAM (up to 2048 MB) in the your coreboot repo, but without success yet.
Could' you help me to make firmware with 2048 VRAM
Thanks

MrChromebox, 2024-03-17:
there's no point in 2GB fixed VRAM on these devices, I will not be making any builds with it set that high

ys202, 2024-03-17:
What about 512Mb or 1Gb? 32Mb seems not enough for normal performance

ys202, 2024-03-19:
> closing this as all PCO boards will have fixed 512MB VRAM going forward

Does this mean that current ROM builds should have 512MB VRAM? 
Should I create a new issue (or you can reopen this one) because the amd software in windows shows only 32Mb VRAM on current ROM build?

MrChromebox, 2024-03-20:
128MB is going to be the amount going forward, starting with the 2402 release

Striker770, 2024-05-18:
> 128MB is going to be the amount going forward, starting with the 2402 release

Been a long time and recently, i thought to give hands on. As the APU is supported by third party kext, the minimum VRAM requirement is 512MB. The custom one you provided, worked well but that time, had crashing due to something else, even if the VRAM was 2048MB. However, if the VRAM is lower than 512MB, the kext won't even load. Tried the latest one and yes, it's 128MB. Is there any way to increase up to 512MB at least starting with your next release for the AMD systems, please? I think 512GB is not too high or too low and for 4GB/8GB systems, it should work well. 

Thanks

MrChromebox, 2024-05-19:
512MB is 12.5% of 4GB, that's too much for those devices IMO. and 128MB fixed is more than sufficient for everyone not trying to run OSX

Striker770, 2024-05-19:
> 512MB is 12.5% of 4GB, that's too much for those devices IMO. and 128MB fixed is more than sufficient for everyone not trying to run OSX

I get your point. What if you can do a custom firmware in private?