MrChromebox/firmware #553: Issues with touchscreen on HP chromebook x360 14c (i3 11th gen)

cdd1000, 2023-11-29:
So I just got my new chromebook and decided I wanted to install RW_LEGACY, because I wanted to keep chromeOS. I have done this before a few times but both times I used the full ROM instead. I set it up and tried a few linux distros (Ubuntu, Manjaro, ElementaryOS) and ended up installing ubuntu. I was able to fix most of the driver issues but I can't get the touchscreen to work. I would really like to be able to use it, is there any driver I can use instead of installing the full UEFI rom? 

Dmesg log attached below:
[https://pastebin.com/pQJNHTRt](url)

MrChromebox, 2023-11-29:
nothing I can do about RWL issues, it's an issue with the stock firmware and differences between the CrOS kernel and mainline linux kernel

MrChromebox, 2023-11-29:
Tigerlake devices are not without their issues under Linux, might want to join the #linux channel on Discord and ask others first

cdd1000, 2023-11-30:
Ok, i will take a look.

closed 2026-11-04 by MrChromebox