MrChromebox/firmware #554: AC power problems on Stoneyridge

GamePlayer-8, 2023-12-04:
Good evening,

I've got a problem with detecting if battery is discharging.
When I unplug the USB-C charging cable from the running Chromebook `HP 11A G6 EE`, system still reports that the battery is charging. Closing lid and opening again makes the system recognize that the battery is discharging.

System: `Artix Linux rolling` with builtin Stoney firmware.

`upower` reports:
 - [when charging](https://gist.github.com/GamePlayer-8/4ecc9865537a297e7f04e4cd3b810076)
 - [when cable is unplugged while running](https://gist.github.com/GamePlayer-8/ab17de42769f259a1134df5bb93dd5d7)
 - [with cable unplugged after reopening lid](https://gist.github.com/GamePlayer-8/bf8cff6074a1e92b90ebf3627d036513)

livexia, 2024-08-27:
Same problem with my Pixelbook Go and fedora 40 kde. 
Closing and reopening the lid also fixes it. 
I also find that rebooting upower.service and plasma-powerdevil.service also fixes it.

After searching someone seems to be able to fix this by [fully discharge the battery](https://forum.manjaro.org/t/battery-icon-is-constantly-in-charging-mode/27605/30) which does not work for me.

livexia, 2024-08-27:
Also the battery life is not affected by this issue, so I think this is a software problem?

closed 2026-11-04 by MrChromebox