MrChromebox/firmware #555: Error mounting boot partition; cannot proceed.

mmayden, 2023-12-07:
When attempting to run Firmware Utility Script I get error that is attached.
[image]

mmayden, 2023-12-07:
Is this related: https://chromium.googlesource.com/chromiumos/docs/+/master/security/noexec_shell_scripts.md

MrChromebox, 2023-12-07:
@mmayden what device, ChromeOS version?

> Is this related: https://chromium.googlesource.com/chromiumos/docs/+/master/security/noexec_shell_scripts.md

not at all

micmalti, 2024-02-03:
Met somewhat of a similar error message on my CB515-2H (Omnigul):
[image]

From `dmesg`: `logical sector size too small for device (logical sector size = 512`
Yet, I checked `/dev/sda12` and it was 64M and empty. I also got the logs:
[image]

Then, following the advice given in [#48](https://github.com/yusefnapora/pixelbook-linux/issues/48), I went ahead and played around with the mount flags but to no avail. I also tried running a ChromeOS recovery but this didn't solve the issue as well.

MrChromebox, 2024-02-03:
this is an issue with the GPT partition layout that the partition cannot be mounted, nothing I can do to fix that. I'll try to find a workaround but may need to get one of these devices on hand for testing

micmalti, 2024-02-03:
What I cannot understand is why this issue is specific to certain Chromebook models. Shouldn't the disk layout be the same across all ChromeOS devices, as described in the [docs](https://chromium.googlesource.com/chromiumos/docs/+/HEAD/disk_format.md)?

micmalti, 2024-02-03:
In [functions.sh](https://raw.githubusercontent.com/MrChromebox/scripts/master/functions.sh), it's stated that `/tmp` isn't used "due to noexec restrictions". So, I changed the script to force `get_cbfstool()` and `get_flashrom()` to use `/tmp` and managed to reach the menu, after changing the mount option of `/tmp` to `exec`:

[image]

Is this a valid workaround?

MrChromebox, 2024-02-03:
@micmalti I've reworked the script to remount /tmp with exec perms and still mount p12 when possible to remove ChromiumOS grub remnants. LMK if it works properly for you now

micmalti, 2024-02-03:
Thanks for the quick fix. It worked. The only error messages that I got were related to `cleanup`, which seems to be WIP:
```
./functions.sh: line 962: syntax error near unexpected token ')'
./firmware.sh: line 1209: cleanup: command not found
```

Looking forward to run the script once Omnigul gets supported. Feel free to contact me for testing.

MrChromebox, 2024-02-04:
> Thanks for the quick fix. It worked. The only error messages that I got were related to `cleanup`, which seems to be WIP:
>
> ```
> ./functions.sh: line 962: syntax error near unexpected token ')' [...]

fixed. I'll add support for it shortly

closed 2024-02-04 by MrChromebox

MrChromebox, 2024-02-04:
so it turns out that even if you remount /tmp with exec and suid etc, you still can't execute programs from there. So I relocated them to /usr/local/bin and everything works now

MrChromebox, 2024-02-04:
Added support / UEFI firmware for OMNIGUL

micmalti, 2024-02-04:
That's great news! Will test it out tomorrow morning with a Manjaro installation. Is there any specific reason why EndevourOS is included in the list of recommended distros but not Manjaro given that it's also Arch-based?

MrChromebox, 2024-02-04:
that's a question for the chrultrabook folks

micmalti, 2024-02-04:
I had second thoughts about proceeding with the UEFI installation because, unlike RW_LEGACY, this will prevent me from reverting back to stock should I wish to do so down the road (current support until 2032). Apologies for not making my intentions clear from the start and for being unable to offer feedback.

MrChromebox, 2024-02-04:
>  this will prevent me from reverting back to stock should I wish to do so down the road

not sure where you got that idea, it's completely incorrect. You can always go back to stock using a backup of your stock firmware.

micmalti, 2024-02-04:
I mistook this line from the [docs](https://docs.chrultrabook.com/docs/firmware/about.html) to mean that way: "Completely removes the ability to run ChromeOS".

MrChromebox, 2024-02-04:
you didn't see the section "reverting to ChromeOS"?
https://docs.chrultrabook.com/docs/reverting/

micmalti, 2024-02-04:
My bad. I'll proceed with the installation.

micmalti, 2024-02-04:
The firmware installation was successful. Is it okay to remove the default partitions?

micmalti, 2024-02-04:
Manjaro is failing to install: `Booting from 'Manjaro' failed: verify it contains a 64-bit UEFI OS`. Attempted twice, once with Ventoy and another without. Unsure whether it's a firmware issue or not until I'll try a different distro.

micmalti, 2024-02-04:
You're right. Did just that. A great thank you for your help.