MrChromebox/firmware #556: EMMC not appear when try install windows 11 on Acer Chromebook Spin 514 AMD Ryzen 3 3250c

metalex10, 2023-12-08:
EMMC not appear when try install windows 11 on Acer Chromebook Spin 514 AMD Ryzen 3 3250c?

Is problem of UEFI or other thing

I also tried to create with windows to go with a micro SD which always works but it doesn't boot from SD either.

Any recommendation?

MrChromebox, 2023-12-08:
known issue, duplicate issue, warned before you even flashed.

closed 2023-12-08 by MrChromebox

socram8888, 2024-01-25:
I'm considering getting an AMD Picasso Chromebook (a Lenovo 300e specificically), since they are really cheap, and found about the eMMC booting issue. I've seen looking through recent Coreboot commits https://github.com/MrChromebox/coreboot/commit/850b6c6254ab34cdbda24818cb0ecca484c01c86, which adds the generic AMD eMMC driver for the Google Trembyle and Dalvoz device trees that all devices extend from.

If I were to compile from source code (given I've looked at https://github.com/MrChromebox/scripts/blob/master/sources.sh and, while it says the date, it doesn't list what the Git head was when they were built so I am unsure if those include it or not), would be booting from eMMC expected to work?

MrChromebox, 2024-01-25:
> would be booting from eMMC expected to work?

no, because it's not a coreboot issue, it's an edk2 one. There is no support for MMIO eMMC/SD devices (which is what Picasso Chromebooks use) in edk2 when used as a coreboot payload currently.

socram8888, 2024-01-25:
Alright, perfect, thanks for your response!

metalex10, 2024-01-25:
> > would be booting from eMMC expected to work?
> 
> no, because it's not a coreboot issue, it's an edk2 one. There is no support for MMIO eMMC/SD devices (which is what Picasso Chromebooks use) in edk2 when used as a coreboot payload currently.

> Alright, perfect, thanks for your response!
I solved my own problem by buying an ngff to m2 converter on Aliexpress, I installed a 256gb m2 and from there the entire installation, I hope this information is useful to you, greetings