MrChromebox/firmware #557: Booting Gallium from UEFI ROM on Acer 311 Spin hangs

Steve3742, 2023-12-08:
Hi
So, I used the firmware-utils script to replace the firmware on my Acer 311 Spin (ApolloLake) and then truied booting from a Gallium 3.1 bood USB. But it just hangs. A curser appeares in the middle of the top lef quadrant and that's it. I've looked at all the UEFI options and it can see the USB, can boot from it, but just hangs. When I try booting from the chromebook, it doesn't boot as there's no OS on it, but does read all 12 sectors and then goes into the UEFI Shell. But from the USB, it just hangs. O tried a different USB and the other USB port, but that made no odds.
I would like to try re-running the firmware-utils again as I wonder if maybe I turned it off before it was complete. Any way of doing that from a Chromebox that can't boot?

Steve3742, 2023-12-08:
Belay that problem, I just tried using a different OS on the USB (Linux Mint 21) and it booted fine. So, the fault is Gallium. Anyone have any ideas or experience as to a resource-light Linux OS that will work on an Acer 311 Spin Chromebook? I think LM 21 might be too resource-heavy.

MrChromebox, 2023-12-08:
GalliumOS is years dead, and never supported your hardware. Pick a modern 2023 distro using a 6.0+ kernel and you'll be fine

closed 2023-12-08 by MrChromebox