MrChromebox/firmware #559: ASUS Chromebox 3 (CN65) TPM support possible?

albertofustinoni, 2023-12-13:
I have a couple of CN60’s and, while at first it was impossible to initialise the TPM, eventually it became usable (got Bitlocker to work).

On the CN65 Windows does not see the TPM at all.

is this something that could be fixed in a future release? I assumed the CN65, being Skylake, would have a TPM 2.0 on board, or an fTPM.

MrChromebox, 2023-12-13:
the CN65 is Kabylake-R. TPM 2.0 is via the CR50 / Google Security Chip. fTPM is fused off from the factory.

CR50 support needs to be added to the UEFI firmware (edk2) so that Windows can see/use it like a normal TPM 2.0. It's a WIP.

mnzaroo, 2023-12-26:
@MrChromebox I need a guide to dualboot chrome os and windows on cn65. I tried brunch it doesn't seem to work. I am currently running windows. Please help.

MrChromebox, 2023-12-26:
I have no such guide. I don't use Brunch nor Windows

PhantomTimed, 2024-03-25:
mnzaroo, just restore stock firmware if you want ChromeOS (unless the device is EOL, which it wouldn't probably work with brunch in the first place if it is)

closed 2026-11-04 by MrChromebox