MrChromebox/firmware #560: Asus Chromebook CX1400CK

IAmSaeve, 2023-12-18:
Looking at the supported list on [https://mrchromebox.tech/#devices](https://mrchromebox.tech/#devices) I see that CX1400FKA.

I assume these are almost the same, but since the documentation specifies that the model number must be exact, I wanted to know if the CX1400CK is supported?

If not is there anyway I could test/check for compatibility without actually trying to flash the BIOS?

Asus product page: [https://www.asus.com/laptops/for-home/chromebook/asus-chromebook-flip-cx1-cx1400/](https://www.asus.com/laptops/for-home/chromebook/asus-chromebook-flip-cx1-cx1400/)

MrChromebox, 2023-12-18:
>  the documentation specifies that the model number must be exact

no, it does not. The documentation is explicitly clear that make/model are irrelevant, and that the ChromeOS board name is the only way to determine if a device is supported or not.

IAmSaeve, 2023-12-18:
So sorry, must have mixed the docs for some other project.

closed 2023-12-18 by IAmSaeve