MrChromebox/firmware #561: Chromebook Pixel 2013 Suspend issue

StackPanda, 2023-12-19:
I recently pulled out my old chromebook pixel in hopes of working on it again and I updated the firmware version to the latest after installing the latest Linux mint and I noticed that the suspend function entireley does not work. I used to use the brunch framework and run the latest chrome os but since then I started wanting to use linux but this is a crazy deal breaker. Is there any way I can fix this? I tried downgrading to the 5.0 kernel but that just resulted in kernel panic at boot :(

P.S. I don't know if anyone still manages this stuff because it is quite old but my apologies if this is relighting another issue.

[image]

StackPanda, 2023-12-19:
Side note, thermals suck. I cannot manage to get a tweak to fix the HUGE issue that to fans do not run to cool the system what so ever, unless I am on the boot screen.

I do not expect that I will get responses on all of these issues but just to put this out there and if anyone has a solution it would be mighty helpful to chime in!

Thank you to Mr. Chromebox for all your work on the firmware and such that has made this possible.

MrChromebox, 2023-12-19:
there's a bug in the cros-ec-lightbar driver that's causing an immediate wakeup, if you blacklist the driver then suspend/resume should work normally -- it does here for me on Pop_OS w/kernel 6.5.6

closed 2023-12-29 by StackPanda