MrChromebox/firmware #563: Bootloop on Asus Chromebook C302CA

thekerker, 2023-12-26:
I installed the MrChromebox UEFI firmware and Fedora Linux on my Mom's Asus Chromebook C302CA (Cave) back in May (MrChromebox 4.20.0 5/013/2023). Everything was running fine until about month ago. The Chromebook now just goes through a boot loop every time it starts up. After flashing the logo, it goes to the Fedora kernel selection screen but no matter which one I choose I'm presented with a blank screen and then it reboots and starts over. After maybe 10 or so tries it'll eventually get to the Fedora login screen but will reboot again before I can even enter the password.

I'm now attempting to install ChromeOS Flex. I created it using the .bin image from Google and Fedora Media Writer on my system. The USB stick is recognized in the Chromebook's boot menu, but like the Fedora image, it goes to a blank screen before just rebooting. Unfortunately it keeps doing this indefinitely instead of taking me to a ChromeOS Flex screen.

thekerker, 2023-12-26:
I also tried booting off of a live Fedora USB image. I see a brief systemd message splash but it disappears and the system reboots before I can read what it says.

MrChromebox, 2023-12-26:
if booting an installed OS and live USB produce the same behavior, then it would appear there is some underlying condition causing the problem. I'm not even sure reflashing the firmware would help (and it's a real PITA to do externally on CAVE)

thekerker, 2023-12-26:
I'm closing the issue as it's not firmware related. Something must have happened with a Fedora update at some point that was causing the boot loop. After multiple attempts I finally got into a live usb of Fedora XFCE and managed to install it. So far no boot loops.

I tried again to install ChromeOS Flex off of a usb, but for whatever reason it just restarts the machine. I'm wondering if the UEFI doesn't like that it's a .bin file, but that's for me to figure out.

thekerker, 2023-12-26:
No issue

closed 2023-12-26 by thekerker

MrChromebox, 2023-12-26:
I do recommend updating to the latest UEFI firmware and clearing NVRAM once you're booted back into Linux

MrChromebox, 2023-12-26:
Flex should be dd'd to the USB just like a linux ISO. Etcher is fine for this as well