MrChromebox/firmware #569: Flashing UEFI ROM with programmer

saspol, 2024-01-17:
Hello, i have Asus Chromebook 2 CN62 motherboard (completely clean, nothing else) and want turn it to normal PC. The default board behaviour was asking for recovery ChromeOS. So i look firmware.sh, get .rom file for this board from mrchromebox website and flash it directly to unsoldered flash chip. After put back motherboard slowly started, CPU get warm but no any picture and so on.
Looks like i miss something, what else i need to do except flashing? This scripts difficult to understand, i did not get clear answer. Please give some advice.

saspol, 2024-01-20:
I update VFD in new firmware as mentioned in unbrick manual and it able to boot. So looks like VFD contains something important, not only MAC/serial. Thank you for reply!

closed 2024-01-20 by saspol

MrChromebox, 2024-01-20:
> I update VPD in new firmware as mentioned in unbrick manual and it able to boot. So looks like VPD contains something important, not only MAC/serial. Thank you for reply!

this is completely incorrect, the VPD is not essential to boot at all. Whatever issue you had was resolved by a reflash, but it was not due to the VPD being present

saspol, 2024-01-21:
I dont think problem with flashing, because i did verifying, etc. Except VPD only is changed is UEFI ROM coreboot_edk2-guado-mrchromebox_20231222.rom -> coreboot_edk2-guado-mrchromebox_20240117.rom

MrChromebox, 2024-01-21:
the VPD is irrelevant. it is read, if present, to populate the device serial # and the LAN MAC Address. If not present, defaults are used. The first 5+ years of MrChromebox releases didn't preserve the VPD