MrChromebox/firmware #570: Pixelbook go running windows

Lkeaheng, 2024-01-19:
[image]

Been running linux for a while without any issue. This week i switched to windows 11 and this happened. 
Im running atlas (pixelbook go) on 4.22.1 firmware. Installed all the updates from windows update.
Turn out the bug is related to the graphic driver. 
Atm i disable the graphic driver and its running fine. 
Sorry for my english
Have a nice day.
[image]

Lkeaheng, 2024-01-19:
[image]

MrChromebox, 2024-01-19:
graphic driver issues aren't something I have any control over. https://forum.chrultrabook.com for Windows support. Try using a driver downloaded from Intel directly, older or newer.

closed 2024-01-19 by MrChromebox