MrChromebox/firmware #571: TPM question on Asus CN60 (panther)

pentastich, 2024-01-20:
Hi! I've been running linux (currently Ubuntu, but I'm switching to Fedora) on an old CN60 for some time, and I just updated to the latest version of your firmware. Thank you!

Since my system uses full disk encryption, I'm exploring using TPM to hold the encryption key. My understanding is that the TPM chip firmware (which is, of course, not YOUR firmware) on my old chromebox has updates available, but it's not clear where to find them, and, if I did, what TPM versions your UEFI software supports.

The firmware EFI boot screen reports that I have an Asus Chromebox CN60 (Panther), and gives me options for TPM 1.2 Configuration.
you
If I boot into linux and run tpm_info, I see:
$ tpm_version
����U  TPM 1.2 Version Info:
  Chip Version:        1.2.4.34
  Spec Level:          2
  Errata Revision:     3
  TPM Vendor ID:       IFX
  Vendor Specific data: 042203f2 0274706d 3338ffff ff
  TPM Version:         01010000
  Manufacturer Info:   49465800

I was wondering whether you're aware of people who have updated the TPM firmware on CN60 systems and, if updated, whether your firmware supports TPM 2.0.

I'm also happy to be told not to bother and to keep typing the encryption password!

Thanks!

Jim

MrChromebox, 2024-01-20:
hi Jim,

TPM updates from Infineon are for a specific part, and typically from version X to Y. They're also not publically available. So to update the TPM firmware, you'd need to find the update file and utility for the specific part and version # you have currently. IMO The only reason to do so would be if the version you have has an exploitable vulnerability that you want to address. While updating from TPM 1.2 to 2.0 might be possible, it would require updated coreboot firmware built specifically for your device with TPM 2.0 enabled (right now only TPM 1.2 is enabled). And I can't promise it would work 100% (or even at all) with the UEFI firmware/config utility as it's not something anyone has done/tested.

pentastich, 2024-01-20:
Thank you! I believe that I know my path forward… do nothing!