MrChromebox/firmware #572: Cant boot any OS

Mrnermaler, 2024-01-23:
I keep getting failed verify its a contains a 64bit UEFI OS but i know it's 64 bit UEFI os because it worked before

MrChromebox, 2024-01-23:
ok? You can't provide a half-assed error description and expect anyone to be able to help. Some basic info would be nice, such as:

* device you're using
* firmware version from UEFI menu
* exact steps you're taking
* full and exact error message (pic is fine)

micmalti, 2024-02-06:
OP probably ran some Linux distro from a USB which, upon installation, failed to boot up from the internal drive (hence the phrase "because it worked before"):

[image]

I'm met with the same problem on my Chromebook Plus 515 (CB515-2H-31NY) with board name `OMNIGUL`. Tried multiple distros (Ubuntu 23.10, EndeavourOS, Manjaro) with and without using Ventoy, but the outcome was the same regardless: the EFI partition showed up in the boot menu (path: `\EFI\[distro name]\grubx64.efi`) but it produced the above error message when selected, followed by this message in the UEFI shell:
`map: No mapping found`

Here's the cbmem logs: [cbmem.log](https://github.com/MrChromebox/firmware/files/14173621/cbmem.log)

MrChromebox, 2024-02-06:
@micmalti boot to the UEFI boot menu. If there's no entry present for the internal storage, then that's your problem -- edk2 doesn't recognize it, and so the 'Manjaro' boot entry points to a device/path that doesn't exist.

micmalti, 2024-02-06:
Meaning Boot Manager → Boot From File like it says in the FAQ? It's empty.

MrChromebox, 2024-02-06:
no, literally press ESC then select Boot Menu. It should list the UEFI shell, the internal storage, and any boot entries added by the OS (eg, 'Manjaro')

micmalti, 2024-02-06:
Yes, there's an entry but it produces the same error message when selected.

micmalti, 2024-02-06:
Shall I consider using efibootmgr as suggested [here](https://superuser.com/questions/1768696/re-adding-efi-file-to-uefi-boot)?

MrChromebox, 2024-02-06:
no. post a pic of your boot menu, as well as the Boot From File options

micmalti, 2024-02-06:
Please note that only one distro is visible in the boot menu since I had previously removed the others:

[image]

[image]

MrChromebox, 2024-02-06:
it's exactly as I said before:

> If there's no entry present for the internal storage, then that's your problem -- edk2 doesn't recognize it, and so the 'Manjaro' boot entry points to a device/path that doesn't exist.

your internal storage is not recognized by edk2, so it's not bootable.

micmalti, 2024-02-06:
Is there a way that I can specify the path manually without using efibootmgr? My understanding is that the solution in the FAQ only works if the EFI partition is already visible in the File Explorer.

MrChromebox, 2024-02-06:
there is no workaround, other than booting from removable media. 

The Manjaro entry already specifies a path -- that path doesn't exist because your internal storage doesn't exist.

edk2 needs to be updated/fixed to support EFS boot media

micmalti, 2024-02-06:
Then it seems I have to revert back, unless there's a fix in the near future. Too bad.

MrChromebox, 2024-02-06:
do you have a SuzyQ cable? that would help in debugging

micmalti, 2024-02-06:
No, but I'll try to find one. I've seen a tutorial on how to make one but I'm not sure that I can do it.

micmalti, 2024-02-06:
I think that in the meantime, it's a good idea to specify in the docs that devices with UFS media are currently unsupported.

micmalti, 2024-02-06:
Where can I obtain an invite?

MrChromebox, 2024-02-06:
https://discord.gg/j5YjPWp9

karmst, 2024-02-16:
My Acer 515 chromebook arrived yesterday, and this issue also affects me. Any idea when the test build will be released?

Thank you.

MrChromebox, 2024-02-16:
there will be no test build released, the solution I tried did not work. Need to get my hands one for testing

karmst, 2024-02-16:
Thank you for the quick reply. Apart from the HD not being detected, I like the machine. For now, I can boot off another USB. Keep up the great work, by the way. I've had Linux running on my older Acer for years.

karmst, 2024-02-17:
I'd be happy to test any builds, if that would help? My intention is never to use the CB as a CB anyway, so I'm not at any loss to rinse and repeat. 

Let me know.

mdrobnak, 2024-04-15:
Is the current workaround to boot off USB, but then the rootfs can still be on a UFS device? Since I had issues with a different device, I have one of these (CB515-2H) on order, because I thought it was supported and inexpensive...and now see this bug. :)

I can certainly deal with the USB workaround for now, and would definitely be up to helping test / attempt to contribute as well. I have another Linux laptop and server that I can set up a dev environment if pointed to some docs.

-Matt

MrChromebox, 2024-04-16:
I've got a fix for UFS-equipped devices, just need to push updates - will try to do so later tonight/tomorrow morning

micmalti, 2024-04-16:
Booting directly from UFS is currently possible, but you need to manually install the missing UFS modules. Something that I've yet to solve is the boot time which takes about 2min (probably a non-critical module that's failing to load). I'll try to test the update and check whether it addresses this issue as well.

mdrobnak, 2024-04-16:
> I've got a fix for UFS-equipped devices, just need to push updates - will try to do so later tonight/tomorrow morning

Awesome! Thanks for all your work.

-Matt

karmst, 2024-04-16:
This is super news! Maybe now I can make use of my Acer Plus Chromebook!  Thank you SO MUCH for your work.

-Keith

MrChromebox, 2024-04-16:
updated 4.22.4 ROMs pushed for all ADL/ADL-N boards

karmst, 2024-04-17:
Hi. I might be doing something wrong, but I updated the firmware to the latest version, and tried to install OpenSuSE and Void Linux, but neither see the UFS disk.

When you say "Booting directly from UFS is currently possible, but you need to manually install the missing UFS modules"...how does one go about that, or is it not related to my issue?

Thank you very much.

-K

micmalti, 2024-04-17:
Just tested Manjaro and Ubuntu 23.10 and both failed to boot out of the box.

micmalti, 2024-04-17:
@karmst, the instructions vary between different distros. The missing module is `ufshcd-pci`.

karmst, 2024-04-17:
@micmalti, thank you for the invaluable information. But, I cannot even get the installer to see the UFS disk. When the installer starts, off a VENTOY disk with ISOs, the disk in the chromebook is not seen. So, I can't even partition the disk to get the OS installed.

Any ideas?

Thank you very much.

MrChromebox, 2024-04-17:
@karmst are you running the latest 4.22.4 firmware (2024/04/16) ? that is mandatory for UFS support

karmst, 2024-04-17:
> @karmst are you running the latest 4.22.4 firmware (2024/04/16) ? that is mandatory for UFS support

Yes, I am. I updated it last night.

MrChromebox, 2024-04-17:
> > @karmst are you running the latest 4.22.4 firmware (2024/04/16) ? that is mandatory for UFS support
> 
> Yes, I am. I updated it last night.

on what device? the UFS storage is not showing under the UEFI boot menu?

karmst, 2024-04-17:
> > > @karmst are you running the latest 4.22.4 firmware (2024/04/16) ? that is mandatory for UFS support
> > 
> > 
> > Yes, I am. I updated it last night.
> 
> on what device? the UFS storage is not showing under the UEFI boot menu?

It's an Acer 515 plus. There is nothing showing in the UEFI boot menu. Ubuntu did see the disk, but then did not boot after install. Void sees the disk for installation, and reports it as 953GB. I can partition it, but then it only recognizes 1 partition, and not the 6 I put on there. If I could get Ubuntu installed, that would be OK.

Weird that the disk does not show up as something in the UEFI menu.

MrChromebox, 2024-04-17:
> Acer 515 plus

board name?

micmalti, 2024-04-17:
You should be able to install Ubuntu 23.10 without any issues. However to boot it, you need to pass a different set of commands in chroot:
`echo "ufshcd" >> /etc/initramfs-tools/modules`
`echo "ufshcd-pci" >> /etc/initramfs-tools/modules`

karmst, 2024-04-17:
I will try ubuntu again after lunch, and boot off the ISO again and redo the grub stuff.

Thanks a million, once again!

-Keith

MrChromebox, 2024-04-17:
> I will try ubuntu again after lunch, and boot off the ISO again and redo the grub stuff.
> 
> Thanks a million, once again!
> 
> -Keith

FFS, none of that will make a difference if the boot device isn't visible in the UEFI boot menu.

WHAT IS THE DEVICE BOARD NAME

karmst, 2024-04-17:
The platform is Intel AlderLake / RaptorLake-U/P.

micmalti, 2024-04-17:
> The platform is Intel AlderLake / RaptorLake-U/P.

Boot with a USB, run MrChromebox's firmware script, and report the BOARD NAME.

karmst, 2024-04-17:
> > The platform is Intel AlderLake / RaptorLake-U/P.
> 
> Boot with a USB, run MrChromebox's firmware script, and report the BOARD NAME.

I just did. Sorry for the delay. The platform is Intel AlderLake / RaptorLake-U/P.

micmalti, 2024-04-17:
That's not the board name. Could you send the terminal info?

karmst, 2024-04-17:
I'll try and get you the information, really I will, but nowhere do I see board name. I run the firmware-util.sh script. Is there something else I can run to get the information?

micmalti, 2024-04-17:
It's `OMNIGUL-FUQB`. You did well :grinning:

MrChromebox, 2024-04-17:
the board name is shown at the top/center of the UEFI settings page, after the device's manufacturer/name, in parentheses. It's also shown in the header of the Firmware Util script, again in parentheses after the device manufacturer/marketing name

MrChromebox, 2024-04-17:
the 4.22.4 release has full support for UFS, so if it's not being listed in the UEFI boot menu, then something else is going on. There is nothing to be done or no workaround from the OS side, it's a firmware issue

karmst, 2024-04-17:
How do I find the point release of the firmware script?  I do see I'm running 2024-04-16, so I would assume it's the latest as I just curled the script following the instructions.

I want to rule out me doing something stupid :)

micmalti, 2024-04-17:
@MrChromebox, @karmst said the model is shown as `OMNIGUL`. I don't see why it can't be bootable if mine has the same board.

karmst, 2024-04-17:
> @MrChromebox, @karmst said the model is shown as `OMNIGUL`. I don't see why it can't be bootable if mine has the same board.

Mine shows nothing in the UEFI boot menu. Nada.

mdrobnak, 2024-04-17:
Stupid question but have you attempted to install an OS yet? You should be able to boot off USB no matter what..
Usually the boot menu will show only items which _are eligible for boot_ ie valid boot sources.

karmst, 2024-04-17:
Yes. I installed Lubuntu 22.04 (not my favourite OS, but just as a test).
Then I did the additional step of adding in the modules and rebuilding the init stuff.
But, I'm out of luck if the UEFI boot menu shows me nothing.

MrChromebox, 2024-04-17:
> Usually the boot menu will show only items which are eligible for boot ie valid boot sources.

no, the internal device will always be listed even if it's not bootable (unless it's not detected in this case, or has manually been removed from the boot device list)

> But, I'm out of luck if the UEFI boot menu shows me nothing.

it means we need to debug further. Do you have a SuzyQ cable?

MrChromebox, 2024-04-17:
> How do I find the point release of the firmware script?

the script doesn't have a version. You can check your fw version on the main page of the UEFI settings, just like with the board name. it literally says, Fw version: xxxxx

mdrobnak, 2024-04-17:
> > Usually the boot menu will show only items which are eligible for boot ie valid boot sources.
> 
> no, the internal device will always be listed even if it's not bootable (unless it's not detected in this case, or has manually been removed from the boot device list)

I stand corrected!

karmst, 2024-04-17:
MMmm...on the UEFI main page, it says the FW is Google OMNIGUL. It that OK, or does that actually mean the FW was not updated?

What should one see on the UEFI page when the FW is updated, as in via the script?

micmalti, 2024-04-17:
@karmst can you send a screenshot of the UEFI boot menu once you've installed Lubuntu?

karmst, 2024-04-17:
[image]
[image]

MrChromebox, 2024-04-17:
<facepalm> you're using RW_LEGACY firmware

[13 more replies]

closed 2024-04-18 by MrChromebox