MrChromebox/firmware #573: ASUS Chromebox curl : (6) Could not resolve host

themanfrommoon, 2024-01-24:
Hello,

I try to install with the line:
cd; curl -LO mrchromebox.tech/firmware-util.sh
But the answer is:
curl : (6) Could not resolve host: mrchromebox.tech

I guess, there is no connection to the internet.
I have connected an ethernet cable and wifi is also available, but I have no idea how to enable/connect it.

Can anyone help me, please?

Thank you and best regards,
Chris

MrChromebox, 2024-01-24:
not sure this is a problem we can help you fix, either the device has network connectivity or it doesn't

themanfrommoon, 2024-01-24:
Any idea what to check and how to solve?

MrChromebox, 2024-01-24:
log in as a guest and open a browser to mrchromebox.tech. try using only LAN or only WiFi. Not sure what to suggest as far as ChromeOS network troubleshooting

themanfrommoon, 2024-01-24:
Problem solved 🙄
I used the wrong LAN port, which was not connected at my switch.
Sorry for for wasting your valuable time 🤐

closed 2024-01-24 by themanfrommoon