MrChromebox/firmware #574: Unable to select TianoCore bootloader on Acer Chromebook (CB311-9H)

leou, 2024-01-24:
After setting Developer Mode, I ran the Firmware Utility Script and loaded the RW_LEGACY firmware.  Then I login as chronos, became superuser with the command `sudo -s`, then ran command `crossystem dev_boot_altfw=1` and rebooted. 
I was not able to select the bootloader shown in the white splash screen.  The screen displays the two lines shown below.

`Press a numeric key to select an alternate bootloader:`

`2 TianoCore    TianoCore bootloader`

I tried pressing the number 2 on the keyboard but nothing happened, pressing Enter did nothing either. There was no visible cursor and the arrow keys did not seem to do anything.

Is this a known issue? My goal is to launch the TianoCore bootloader so that I can boot from a USB drive to then install Arch Linux.

I'd appreciate any help in resolving this issue.
Leo

MrChromebox, 2024-01-25:
> then ran command crossystem dev_boot_altfw=1 and rebooted.

why? where in my instructions did it say to do that?

>Is this a known issue?

yes, and it's documented on the supported devices page . If you're getting a menu, you didn't update the RW_LEGACY firmware.

> My goal is to launch the TianoCore bootloader so that I can boot from a USB drive to then install Arch Linux.

if your goal is to replace ChromeOS with Arch, then you want the UEFI Full ROM firmware; RW_LEGACY is for dual booting

leou, 2024-01-25:
I was following instructions in an Arch wiki page. I thought that RW_LEGACY was a valid option and it was the only option I could choose when I ran the Firmware Utility. I don't know how to disable the firmware write protection in order to do UEFI Full ROM update.

Thanks for the clarification.

leou, 2024-01-25:
Arch wiki page: https://wiki.archlinux.org/title/Chrome_OS_devices

The only option I could select when I ran the Firmware Utility was: 1) Install/Update RW_LEGACY Firmware

The entry in the Supported Devices page notes that RW_LEGACY and Full ROM are supported and that the WP method is CR50 (battery). I don't have a Suzy-Q cable so I guess the only way to disable WP is by disconnecting the battery. I'll have to find out if/how I can do this on this CB311 model.

MrChromebox, 2024-01-25:
> The only option I could select when I ran the Firmware Utility was: 1) Install/Update RW_LEGACY Firmware

before doing anything else, please take a picture of the main menu when running my script and post it here

leou, 2024-01-25:
Here is the picture.
[image]

leou, 2024-01-26:
Thanks for your help. I did as suggested above. Full ROM was installed and Arch is up and running!

Before flashing the ROM the script  created a backup in a USB drive. Is this in case I want to go back to ChromeOS? How is this backup used? Any documentation?
Leo

MrChromebox, 2024-01-26:
>  Is this in case I want to go back to ChromeOS? How is this backup used? Any documentation?

yes. run the script from linux, choose option to restore stock firmware, select option to restore from USB backup. Documented on my site.

closed 2024-02-15 by MrChromebox