MrChromebox/firmware #575: byra / redrix : missing Intel VT?

aoberoi, 2024-01-25:
first of all, thanks for this project. it's an incredible undertaking and it enables so much good. also, i'm a noob when it comes to firmware things, but i'd like to learn, so i appreciate your patience.

I have MrChromebox-4.22.2 installed on a Redrix. I don't see any options in the boot menu mentioning virtualization being enabled or disabled. Based on reading previous issues, its my understanding that the firmware installation scripts would enable virtualization if the machine supported it. However, this machine has the Intel i7-1265U CPU, which as far as I can tell, [does support various Intel VT technologies](https://www.intel.com/content/www/us/en/products/sku/226258/intel-core-i71265u-processor-12m-cache-up-to-4-80-ghz/specifications.html). is my understanding incorrect? if not, is there anything i can do to help land a change to enable the virtualization option in the firmware?

for background: i have Windows 11 installed, and i was trying to set up WSL2. it seems to have failed because I get a BSOD with the stop code `INACCESSIBLE_BOOT_DEVICE` on reboot. booting into Safe Mode seems to revert the change and then I can boot up normally. when i tried to investigate, my only lead was part of the output from the `wsl --status` command:

```
WSL2 is not supported with your current machine configuration.
Please enable the "Virtual Machine Platform" optional component and ensure virtualization is enabled in the BIOS.
Enable "Virtual Machine Platform" by running: wsl.exe --install --no-distribution
For information please visit https://aka.ms/enablevirtualization
```

this led me to look into BIOS settings and realize that virtualization wasn't present.

MrChromebox, 2024-01-25:
> this led me to look into BIOS settings and realize that virtualization wasn't present.

there is no setting because it's not user configurable. VT-x and VT-d are both enabled on all parts which support it. You can verify this by looking at the CPU info on the Performance tab in Task Manager, or running CPU-z, or any other number of utilities. Your issue is not a firmware setting.

aoberoi, 2024-01-26:
thanks for the quick response.

as you mentioned, i can see virtualization is enabled. in CPU-z, i see VT-x, but not VT-d. when you say these are not user configurable, does that mean the hardware is locked at VT-x enabled and VT-d disabled? i've been googling around and all i've seen is people sharing screenshots of where its configured in their BIOS, i hadn't considered that my hardware was not configurable.

MrChromebox, 2024-01-26:
> when you say these are not user configurable, does that mean the hardware is locked at VT-x enabled and VT-d disabled?

it means that coreboot automatically enables those items if the hardware supports them, and there are no user options to enable/disable any features like you might find in a proprietary UEFI firmware. if your CPU supports

> all i've seen is people sharing screenshots of where its configured in their UEFI/BIOS, i hadn't considered that my hardware was not configurable.

you're not running anything close to the same firmware, what others can do is irrelevant. The hardware is configurable. The firmware is not (not without rebuilding).

All that said, I'll double check and make sure that the Alderlake SoC code is correctly setting and locking the VMX register bits

coolstar, 2024-01-26:
WSL2 and VMWare run fine on brya / banshee here. You'll need to search in start menu for "turn windows features on or off" and enable "Virtual Machine Platform"

aoberoi, 2024-01-27:
> it means that coreboot automatically enables those items if the hardware supports them, and there are no user options to enable/disable any features like you might find in a proprietary UEFI firmware. 

i understand that there are currently no user options. my question is whether or not user options can be added? if so, in which project(s) would that be done?

MrChromebox, 2024-01-27:
> my question is whether or not user options can be added (in theory, at least)? if so, in which project(s) would that be done?

I'm not going to add pointless options. There's no need ever to disable VT-x or VT-d, and I have no desire for coreboot to look or act like some POS proprietary UEFI firmware. 

Adding any firmware config options requires doing so in both coreboot and edk2, and having some way to tie them together.

aoberoi, 2024-01-30:
Okay, that's reasonable, thanks for considering.

My use case is simply to try out this workaround to what seems like a popular issue for Windows users (not limited to Chromebox firmware, but overlapping). Simply a vote in favor of some more configuration. Thanks!

closed 2024-01-30 by aoberoi

xiaojianqaq, 2026-02-07:
@aoberoi
I had the same issue in Windows 11 23H2, but it was resolved after I switched to the official 25H2 build. The system no longer crashes to a BSOD and boots normally. I hope this helps you even after all this time.