MrChromebox/firmware #579: GPU in Asus chromebox 3??

PuravIPatel, 2024-02-07:
Hello, I have tried to play some games in Asus chrome box 3,it have i7-8550u,and igpu, but it is not sufficient for new games like gta 5 and more.
so I have tried to install gpu with m.2 to pice converter and  i did't get display from gpu.
booting in windows I got this in device manager...
I have connected gtx1650 4gb.

need help.....
[image]
[image]
[image]
[image]
[image]
[image]

m.2 to pice 
https://robu.in/product/black-new-high-quality-ngff-m-2-to-pci-e-4x-1x-slot-riser-card-adapter-m2-to-pcie-x4-x1-converter-for-graphics-card/

MrChromebox, 2024-02-07:
please use https://forum.chrultrabook.com for Windows issues, you've clearly not installed all the necessary drivers for the device. 

As for using an external GPU, I'm not sure this is a supported configuration for this device.

tirsojrp, 2024-04-22:
That board have some issues detecting PCIe devices. https://github.com/MrChromebox/firmware/issues/512

IMHO you are stretching your luck with that setup.