MrChromebox/firmware #580: HP Elemi (Pro c640) doesn't turn on after flashing full ROM and no shellball found

teammonoxide, 2024-02-19:
Hi! Have a HP Pro c640 g2 (ELEMI) which doesn't turn on after full rom flashing with "coreboot_edk2-elemi-mrchromebox_20240120.rom". Also, there's no shellball available ("shellball.elemi.bin")

Is there a older rom available?

MrChromebox, 2024-02-19:
I don't provide shellballs anymore, too much of a headache. You can flash the backup you made before flashing the UEFI firmware, or you can extract a shellball image using the crosfirmware.sh script: https://github.com/coreboot/coreboot/tree/main/util/chromeos

teammonoxide, 2024-02-19:
Oh okay! But aint interested in comeback to chrome's functions. I want to convert to a better life as a normal PC :D

MrChromebox, 2024-02-19:
do you have a SuzyQ cable? If so, then I can give you a debug build to flash and we can see what the issue is

vrdevelopersco, 2024-02-20:
Unfortunately I don't have a SuzyQ cable. Can't be flashed via flashrom ways?

MrChromebox, 2024-02-20:
flashrom is the software part, the SuzyQ is the hardware part. If the flash chip is a SOIC-8 then you can use a chip clip and a ch341a USB flasher, but a SuzyQ cable/adapter would be more useful since you can get debugging output from it as well.

https://www.ebay.com/itm/335130747039

jpr233, 2024-03-30:
hi,
I have Elemi also via legacy rw rom , 
Is full UEFi rom still corrupted  after  an january-february hosting site corruption ?
Headache is also for user reading the compatibility list ... :) thx

vrdevelopersco, 2024-03-30:
> hi, I have Elemi also via legacy rw rom , Is full UEFi rom still corrupted after an january-february hosting site corruption ? Headache is also for user reading the compatibility list ... :) thx

What you can and you can't in RW LEGACY rom?

MrChromebox, 2024-03-30:
> Is full UEFi rom still corrupted after an january-february hosting site corruption ?

it never was

> Headache is also for user reading the compatibility list ... :) thx

what do you mean?

jpr233, 2024-03-31:
> > hi, I have Elemi also via legacy rw rom , Is full UEFi rom still corrupted after an january-february hosting site corruption ? Headache is also for user reading the compatibility list ... :) thx
> 
> What you can and you can't in RW LEGACY rom?

I can't have linux kernel ' elants_i2c kernel built-in touchscreen for Elan (31aa product, 5917 fw)  working on it even  with latest unbuntu 23-03-2024 k 6.8 with k6.8 overwriting dmesg log with 5000/s error messages... till unbind, but further bind making ts working...

I was also asking if UEFI rom that bricked Elemi chromebook was now repaired, it seems that NO...

jpr233, 2024-03-31:
> > Is full UEFi rom still corrupted after an january-february hosting site corruption ?
> 
> it never was
=====> of course YES IT WAS ! 
> 
> > Headache is also for user reading the compatibility list ... :) thx
> 
> what do you mean?
Headache is not only for you (as you previously answered in that forum to do not care of previous legit remark) but for people thinking your legacy or full EFI was an hope to get unix full working on their chormebook..... 
20 years  later my first tries, Windows seems to be finally the only OS to work oob 100% on hw. 
Shame on linux kernel maintainers and you, also, MrChromebook, for all your mistaken hopes. 
Compatibility list was said to be full compatibility on Elemi, without any restriction.. and it's not the true case, as touchscreen  is not working on different linux distros at march,31 2024 where elants_i2c could be faulty because of your acpi rom parameters (RW legacy). I do not dare to try Full rom EFI fearing bricked laptop you did'nt explain.

jpr233, 2024-03-31:
I'm 40 years in IT. Tried 20 years of linux distros.  Till on windows... as I see no challenger, & that make me ill with all these years of hope.... MX & Zorin are my best tries, but chromebooks are somewhat touchscreen exception, while Mr Liu elants_i2c the 'builtin' lernel module (no other touch something are built in, only load modules ...)  means that driver should be the MUST.
 Not at all ! : if you do not belong a kernel compile platform, you are stucked.... Elants_i2c is a weird code, but mrchromebox site tells you your chromebook is 100% compatible. !
Lyers tribu, incredible message that you can probe swinging to linux distros. 
I cannot ack all that mistales years after years. Tell the truth now !

MrChromebox, 2024-03-31:
> > > Is full UEFi rom still corrupted after an january-february hosting site corruption ?
> > 
> > 
> > it never was
> > =====> of course YES IT WAS !

The files were briefly unavailable for ~2 days while I moved to a new host. There was never any corruption.

> Headache is also for user reading the compatibility list ... :) thx
>
>
> what do you mean?
> Headache is not only for you (as you previously answered in that forum to do not care of previous legit [...]

I explicitly do not provide information on OS compatibility, I only provide a list of firmware availability by type. The Chrultrabook project handles OS compatibility.

RW_LEGACY firmware is just a bootloader. You're using the ChromeOS firmware hardware init, ACPI, etc. 

ELEMI having issues under Linux is not a firmware problem.

> but mrchromebox site tells you that your chromebook is 100% compatible

sorry, but your lack of reading comprehension is not my problem.

closed 2024-03-31 by MrChromebox