MrChromebox/firmware #581: Windows11 BSOD after wakeup from sleep

megacar1, 2024-02-21:
Sparky360 machine BSODs after wakeup from sleep (not every time, 5 out of 10) in Windows11
(crash does not happen under linux)

windows logs suggest to look for new firmware

```
error, event 137, kernel-power: The system firmware has changed the processor's memory type range registers (MTRRs) across a sleep state transition (S4). This can result in reduced resume performance.
error, event 34, kernel-processor-power: Idle power management features on processor 0 in group 0 are disabled due to a firmware problem. Check with the computer manufacturer for updated firmware.
error, event 34, kernel-processor-power: Idle power management features on processor 1 in group 0 are disabled due to a firmware problem. Check with the computer manufacturer for updated firmware.
error, event 34, kernel-processor-power: Idle power management features on processor 2 in group 0 are disabled due to a firmware problem. Check with the computer manufacturer for updated firmware.
error, event 34, kernel-processor-power: Idle power management features on processor 3 in group 0 are disabled due to a firmware problem. Check with the computer manufacturer for updated firmware.
warning, event 219, kernel-PnP: The driver \Driver\WudfRd failed to load for the device HID\GOOG0006\7&32a0ea0&0&0000.
warning, event 219, kernel-PnP: The driver \Driver\WUDFRd failed to load for the device ACPI\INT3400\0.
```

megacar1, 2024-02-21:
i also have 3 devices without drivers.
do you think some of these could be the culprit?

unknown device on Intel(R) Serial IO I2C Host Controller - 31AC
ACPI\PRP0001\4&2C41410E&0

unknown device on Intel(R) Serial IO SPI Host Controller - 31C2
ACPI\PRP0001\0

PCI Data Acquisition and Signal Processing Controller on PCI bus 0, device 0, function 1
PCI\VEN_8086&DEV_318C&SUBSYS_318C8086&REV_03\3&B1BFB68&0&01

Mrnermaler, 2024-02-22:
Coolstarorganization@gmail.com

megacar1, 2024-04-22:
hi, updated few weeks ago, much better, since then only 1 or 2 BSOD with driver_irql_less_or_equal

MrChromebox, 2024-04-22:
closing as this isn't a firmware issue

closed 2024-04-22 by MrChromebox