MrChromebox/firmware #585: SCHCI REGISTER DUMP, Lenovo Yogo11e Intel Braswell ULTIMA

HighOnVoltage, 2024-02-27:
Hi mad lads,

Although I've managed to install coreboot via MrChromebox script, doesn't matter the Linux OS I always seem to have this issue upon installation. Tried Debian, Manjaro, Arch & Ubuntu with all the same problems upon installing data to the MMC drive. The live distros work and I can even use gparted on the MMC drive to wipe without fail. 

My EFI 2 boot partitions are only 4mb each and my understanding reading the Arch wiki I most likely need a 512MB FAT EFI boot partition. Done heaps of research of others with a similar issue and nothing was consultive. Even updated to the lastest EFI rom from MrChromebox! 

I've also tried several different brands of SD & USB drives to install media thinking one was corrupt, yet this isn't the case. The result is always the same when looking at my dmesg output. 

Unsure how I've screwed this up, but would be delighted in any input. 
Many thanks,

-- Trev

Could a wise dev please help a newbie chromebook linux converter out? EFI is still very new to me.
[image]
Only have screengrabs due to the nature of issue.

Artox, 2024-07-02:
> this seems to be an issue with the UEFI Full ROM firmware, and I think there might be a kernel command line workaround you can use, but don't have it handy. Might need to dump the registers when booted from stock firmware + RW_LEGACY to see what's different

Hi, I might be motivated enough to make this experiment  and dump  the registers.
Is there a trick to do so using sysfs - or does it require compiling custom kernel binary?

MrChromebox, 2024-07-03:
I don't know, I was just speculating generally. `sudo lspci -xxx -s <pci bus:device.function>` might suffice

Artox, 2024-07-03:
I believe on this Chromebook SDHC is not a PCI device.

For future reference, this is "lspci -vvv" on the original chromeos, latest (end of life) version:

```
localhost ~ # lspci -vvv
00:00.0 Host bridge: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series SoC Transaction Register (rev 21)
        Subsystem: Intel Corporation Device 7270
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Kernel driver in use: iosf_mbi_pci

00:02.0 VGA compatible controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Integrated Graphics Controller (rev 21) (prog-if 00 [VGA controller])
        Subsystem: Intel Corporation Device 7270
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 312
        Region 0: Memory at d0000000 (64-bit, non-prefetchable) [size=16M]
        Region 2: Memory at c0000000 (64-bit, prefetchable) [size=256M]
        Region 4: I/O ports at 1000 [size=64]
        Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
        Capabilities: [d0] Power Management version 2
                Flags: PMEClk- DSI+ D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [90] MSI: Enable+ Count=1/1 Maskable- 64bit-
                Address: fee04004  Data: 4024
        Capabilities: [b0] Vendor Specific Information: Len=07 <?>
        Kernel driver in use: i915

00:0b.0 Signal processing controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series Power Management Controller (rev 21)
        Subsystem: Intel Corporation Device 7270
        Control: I/O- Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Interrupt: pin A routed to IRQ 0
        Region 0: Memory at d131c000 (64-bit, non-prefetchable) [size=4K]
        Capabilities: [40] MSI: Enable- Count=1/1 Maskable- 64bit-
                Address: 00000000  Data: 0000
        Capabilities: [50] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Kernel driver in use: proc_thermal

00:14.0 USB controller: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series USB xHCI Controller (rev 21) (prog-if 30 [XHCI])
        Subsystem: Intel Corporation Device 7270
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B+ ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 313
        Region 0: Memory at d1300000 (64-bit, non-prefetchable) [size=64K]
        Capabilities: [70] Power Management version 2
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0-,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [80] MSI: Enable+ Count=1/8 Maskable- 64bit+
                Address: 00000000fee08004  Data: 4024
        Kernel driver in use: xhci_hcd

00:1b.0 Audio device: Intel Corporation Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series High Definition Audio Controller (rev 21)
        Subsystem: Intel Corporation Device 7270
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 315
        Region 0: Memory at d1314000 (64-bit, non-prefetchable) [size=16K]
        Capabilities: [50] Power Management version 2
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=55mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [60] MSI: Enable+ Count=1/1 Maskable- 64bit+
                Address: 00000000fee02004  Data: 4026
[... 163 lines cut ...]
                LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete- EqualizationPhase1-
                         EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
                         Retimer- 2Retimers- CrosslinkRes: unsupported
        Capabilities: [100 v1] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP+ BadDLLP- Rollover- Timeout- AdvNonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
[... cut ...]
                Max no snoop latency: 0ns
        Capabilities: [154 v1] L1 PM Substates
                L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
                          PortCommonModeRestoreTime=30us PortTPowerOnTime=60us
                L1SubCtl1: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+
                           T_CommonMode=0us LTR1.2_Threshold=163840ns
                L1SubCtl2: T_PwrOn=60us
        Kernel driver in use: iwlwifi
        Kernel modules: iwlwifi
```

MrChromebox, 2024-07-04:
> I believe on this Chromebook SDHC is not a PCI device.
> 
> For future reference, this is "lspci -vvv" on the original chromeos, latest (end of life) version:

that doesn't help, since the eMMC controller isn't in PCI mode with the stock ChromeOS firmware (it uses ACPI mode instead) . It is a PCI device under upstream coreboot / MrChromebox firmware -- it's device 00:10.0 on all Braswell Chromebooks

Artox, 2024-07-04:
> this seems to be an issue with the UEFI Full ROM firmware, and I think there might be a kernel command line workaround you can use, but don't have it handy. Might need to dump the registers when booted from stock firmware + RW_LEGACY to see what's different

Right. So I currently have stock firmware + seabios - which as you say is not exposing sdhci controller by pci.
The question is do we have a way of dumping sdhc registers to compare later with uefi?

MrChromebox, 2024-07-04:
> Right. So I currently have stock firmware + seabios - which as you say is not exposing sdhci controller by pci. The question is do we have a way of dumping sdhc registers to compare later with uefi?

you'd have to dump starting at the MMIO address where the eMMC controller is mapped in memory

kquote03, 2024-08-04:
> yes, appears to be the same issue of the kernel enabling command queuing when the hardware doesn't support it / is broken for some reason.

I tried `sdhci.debug_quirks=0x65168080` (which did nothing), `sdhci.debug_quirks=0x2020000` (which ended up disabling the entire eMMC) and `sdhci.debug_quirks=0x20000` (which also did nothing, despite corresponding to SDHCI_QUIRK_BROKEN_CQE according to sdhci.h)

Sadly nothing worked, would the problem lie elsewhere? Thank you for your time

MrChromebox, 2024-08-05:
sadly I don't have any other ideas currently

polypolyman, 2025-04-29:
I just got one of these ULTIMA devices to play around with, and started running into this same issue on any recent Linux. I may have found a workaround.

Turns out, this has nothing to do with CQE, as this hardware doesn't even give CQE as capable in caps2 (so behavior is identical to debug_quirks=0x20000 anyway). However, there's clearly an interrupt timing issue (the traces I see are all either timeout waiting for interrupt or interrupt without request - we timeout, then finally a bit later the interrupt comes, but we've given up that request). This made me suspect DMA - and sure enough, disabling both ADMA and DMA (debug_quirks=0x60) works (I did test just disabling ADMA, debug_quirks=0x40, and this was not enough. I never tested debug_quirks=0x20). Interestingly, in issue #310 the idea of debug_quirks=0x40 came up, but never 0x60. With those settings, dmesg reports the mmc controllers coming up in PIO mode as expected.

I'm still testing this, so I'll update if I run into any issues, but so far so good, and the performance hit isn't as bad as I may have suspected. I don't understand the nitty-gritty too well at this point, so it's possible that this is more about rate-limiting requests rather than broken DMA, but I wonder if anyone else is able to use this as a functional workaround for now.

refutationalist, 2026-02-10:
So, I have a pile of these I'd like to use for a project, but PIO for both eMMC and the SD slot kind of has me stopped in my tracks with read speeds in PIO mode being rather painful.    Is there anything I could be poking at to find a resolution?  Quirks alone doesn't seem to be finding a way through.

refutationalist, 2026-02-15:
Oh dang, thought you had one.    I'll contact you.

refutationalist, 2026-05-28:
> if you're in the US, sending me one for testing would be a huge help, since I can dump the registers and see what's different between the stock firmware and mine.

System mailed your way.  Did you receive it?

MrChromebox, 2026-09-09:
https://github.com/MrChromebox/firmware/issues/310#issuecomment-5604088999

MrChromebox, 2026-09-27:
fixed in 2609.0 release

closed 2026-09-27 by MrChromebox

Artox, 2026-10-24:
> 24\. fixed in 2609.0 release

Amazing!