MrChromebox/firmware #589: Add support to VELL Chromebook

gabriel-lando, 2024-03-09:
Hello, I have a VELL Chromebook.
Based on the coreboot documentation, this model should be supported, but when I try to run the `firmware-util.sh` script, I receive this error message:

```
Device: ERROR: unknown or unidentifiable device (VELL)
Platform: (unrecognized)
Fw Type: Stock ChromeOS w/RW_LEGACY
Fw Ver: Google_Vell.14505.485.0 (05/23/2023)
Fw WP: Disabled
``` 

Is it possible to add support to this platform? I tried to build by myself the coreboot with edk2 for this platform, but I couldn't succeed.
Is there any tutorial that I can follow to build and install a custom firmware?

Thanks

MrChromebox, 2024-03-09:
> Based on the coreboot documentation, this model should be supported

the MrChromebox firmware distribution does not support every board that the coreboot project does. It tries to support all x86_64 ChromeOS devices. New boards are released all of the time, and if based on an older platform, support isn't added until someone requests it. VELL was released well after the rest of the Alderlake-based boards.

> Is it possible to add support to this platform? 

done

> I tried to build by myself the coreboot with edk2 for this platform, but I couldn't succeed.

can't help you there without knowing what the problem was

> Is there any tutorial that I can follow to build and install a custom firmware?

https://docs/chrultrabook.com might have something

closed 2024-03-09 by MrChromebox

gabriel-lando, 2024-03-09:
It worked :) Thanks a lot

junnikokuki, 2024-03-18:
> It worked :) Thanks a lot

What os do you use? Linux or Windows?

gabriel-lando, 2024-09-20:
> > It worked :) Thanks a lot
> 
> What os do you use? Linux or Windows?

Sorry for the late response :D I'm using Windows. The OS works well, it automatically installed almost all drivers. The firmware has some issues with C-states, so sometimes the hibernation doesn't work well, but except for that, everything works fine

migandro, 2026-03-18:
> Hi. Yeah, webcam and microphone are not working at all. Speaker works on Ubuntu, but not on Windows. Screen is pretty bright on max brightness, the brightness can be adjusted on Windows and Ubuntu.
> 
> As I use it more like a testing machine, these issues don't affect me. Also, I can help to debug if you guys want :)

Thanks so much for getting back to me and sharing these details - information is thin on the ground for this model! Could you share which versions of Ubuntu and Windows you're using? Also, does the trackpad work in both? And does the USB-C/Thunderbolt work in Ubuntu (I saw I could buy the paid driver for Windows)?