MrChromebox/firmware #592: ASUS Chromebox 3 / CN65 (Teemo): USB-C power/HDMI not working

vm75, 2024-03-12:
Device: ASUS Chromebox 3 / CN65 (Teemo)
CPU:Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz
Memory: 16G DDR4 2400 at DIMM0
Harddisk: 512G NVME SSD
UEFI firmware: MrChromebox-4.22.2-5-g822782a1001
Flash the BIOS chip by using the script.

With stock firmware, I was able to use an adapter to provide power and connect HDMI/peripherals using a USB-C adapter. 

With coreboot firmware, it blinks the red LED and does not power up.
I can power using a barrel-jack adapter, but HDMI over USB-C doesn't work.

vm75, 2024-03-12:
Is there any workaround for this?
Thanks

MrChromebox, 2024-03-12:
> I can power using a barrel-jack adapter, but HDMI over USB-C doesn't work.

what do you mean doesn't work? BJ for power + USB-C to monitor works fine here

vm75, 2024-03-12:
I have two USB-C adapters, including [Anker](https://a.co/d/8r8jxZn). I thought it used to work as well initially, but since I installed Windows, even boot screen does not show via the adapter. If I connect HDMI directly, it works.

vm75, 2024-03-12:
Will the power over USB-C work if the TDP is reduced? I saw another issue about that.

MrChromebox, 2024-03-12:
> I have two USB-C adapters, including [Anker](https://a.co/d/8r8jxZn). I thought it used to work as well initially, but since I installed Windows, even boot screen does not show via the adapter. If I connect HDMI directly, it works.

I tested with a direct USB-C connection as well as a UNI brand USB-C to HDMI adapter, both work for the firmware boot screen and Linux, don't have Windows handy to test

> Will the power over USB-C work if the TDP is reduced? I saw another issue about that https://github.com/MrChromebox/firmware/issues/242

the two have nothing to do with one another

vm75, 2024-03-12:
> the two have nothing to do with one another

I meant the power delivery via USB-C, if supported.
I am not quite sure what is happening with the HDMI. I can try re-flashing

bam80, 2024-03-16:
Might be related:
https://github.com/MrChromebox/scripts/issues/186

vm75, 2024-03-18:
that is it. if I remove the screw, it works

closed 2024-03-18 by vm75