MrChromebox/firmware #593: Acer chromebase ca24v2 black screen after firmware flash

mugi21, 2024-03-15:
Hi, I have an acer chromebase ca24v2 (Board name: Karma) that I am trying to convert over to Windows. I flashed the firmware following the instructions on MrChromebox's website. 
After rebooting I got a black screen, fortunately I have a backup of the original firmware which I flashed with an eprom programmer. The device is back to normal.
Can anyone help me with this issue? Thanks a lot.

MrChromebox, 2024-03-16:
can you reflash again and see if issue persists? or reflash with the external programmer? KARMA has been tested working by others

MrChromebox, 2024-03-16:
has nothing to do with enterprise enrollment, flashing the UEFI firmware completely bypasses that.

try flashing this slightly older version and LMK if it works.

Do you have a SuzyQ cable to be able to get serial output from a debug build if I send one?

[coreboot_edk2-karma-mrchromebox_20231222.zip](https://github.com/MrChromebox/firmware/files/14624122/coreboot_edk2-karma-mrchromebox_20231222.zip)

mugi21, 2024-03-16:
Thanks, I'll try that and get back to you.
Unfortunately I don't have a SuzyQ cable.

mugi21, 2024-03-17:
No luck! Still same issue

MrChromebox, 2024-03-17:
that's bizarre as no one else has reported an issue. Let's go back a few more versions

[coreboot_edk2-karma-mrchromebox_20230721.zip](https://github.com/MrChromebox/firmware/files/14627837/coreboot_edk2-karma-mrchromebox_20230721.zip)

mugi21, 2024-03-17:
it didn't work, I start thinking that the issue is specific to my device as no one else has reported such issue.

gourdcaptain, 2024-03-18:
Having similar issues of a black screen after UEFI Full ROM flash on a Dell Latitude 5430 Chromebook (CROTA)...

MrChromebox, 2024-03-19:
@gourdcaptain whatever issue is affecting your board is going to be completely unrelated, the hardware is completely different

mugi21, 2024-03-28:
I tried with another same model device and had same issue: black screen no backlight.

MrChromebox, 2024-03-29:
someone with a KARMA device and a SuzyQ cable will need to flash a debug build to help identify the issue

Sekr070, 2024-04-07:
I had a similar problem after the Acer Chromebase CA24V2 (Board Name: Karma) firmware. The image was displayed only through the HDMI connector. I connected the external monitor and installed Windows and drivers. After Windows loaded, the image began to be displayed on the inner screen. Before loading Windows, the image is not displayed in order to go to UEFI you need to connect an external monitor.

MrChromebox, 2024-04-07:
@Sekr070 please get a cbmem console log from Windows or Linux for me, so I can see what's going on with the pre-OS internal display init

Sekr070, 2024-04-10:
Tell me how to do this. I have Windows 10 installed.

Sekr070, 2024-04-12:
I ran the command  .\cbmem.exe -1 > cbmem.log . What should you do next?

MrChromebox, 2024-04-12:
open the log to ensure it's valid, then attach here as a file. 

DO NOT COPY/PASTE IT INTO HERE

Sekr070, 2024-04-12:
where can I find this log?

MrChromebox, 2024-04-12:
did you run the command from the directory containing the cbmem.exe file? that's required, and where the log will be

Sekr070, 2024-04-13:
[cbmem.log](https://github.com/MrChromebox/firmware/files/14967174/cbmem.log)

MrChromebox, 2024-04-13:
I pushed an updated image for KARMA, update via my script under Linux and see if that fixes the issue

Sekr070, 2024-04-14:
[image]
[image]
The image appeared, but it was upside down.

MrChromebox, 2024-04-14:
ok, pushed another update

Sekr070, 2024-04-15:
Everything works correctly. Thank you so much for your work!

tamilnilan, 2025-09-27:
acer chromebase ca24v2 posible windows instal