MrChromebox/firmware #594: Can't boot fedora 39 from an SD card

BlueIce1390, 2024-03-16:
Im using an HP chromebook 11 G9 EE , and i installed fedora 39 onto an sd card (not a live installation, the full OS)
for some reason it does not appear in the boot menu
Anyone know a fix for this

GalacticZXS, 2024-10-09:
There is actually NOTHING you can do about this, I have a couple of these chromebooks and none of them can boot from an SD card. So, a USB Flash Drive is your best bet, just make sure to have at least 8gb on it so you don't have to find a compatible OS that fits in anything under 8gb.

MrChromebox, 2024-10-09:
generally speaking, Chromebooks with a SD card reader connected directly to the PCI bus don't work, while those connected via USB do

closed 2026-11-04 by MrChromebox