MrChromebox/firmware #595: Asus Chromebox 3: can't boot from USB-C Power Delivery after flashing UEFI firmware

bam80, 2024-03-16:
First of all it still boots from power adapter.
Then, if I swap to PD with WP screw in place, LED blinking red and it doesn't boot.
If I remove the screw, it bootloops.

With original firmware, I can boot from PD with screw in place only. If I remove the screw, get bootloop (need to detach from power before removing the screw, otherwise it can still boot).

So it seems no way of using native Linux with PD currently, which is a whole point I bought the device.

Any help? @MrChromebox could you try to reproduce, please?

bam80, 2024-03-18:
> closing as this isn't a script issue

Thank you for the transferring, I suppose we can reopen it now?

Soon I'll have Suzy-Q cable so can provide more data hopefully.

MrChromebox, 2024-03-19:
all 3 CN65s I have here boot via USB-C PD and provide video via USB-C PD

MrChromebox, 2024-03-19:
is it a coreboot issue? I suspect not. would need to compare ECRW builds between stock and my fw to see if there's something in the ECRW update causing a problem

bam80, 2024-03-19:
Thank you for reopening.

With the original firmware, I can boot at least somehow:
> With original firmware, I can boot from PD with screw in place only. If I remove the screw, get bootloop (need to detach from power before removing the screw, otherwise it can still boot).

I didn't update for a long time though, but will return to all this soon.

BTW, if you remind me how could I get fresh original Google's firmware for the test, that would be helpful.
Thanks.

bam80, 2024-03-19:
> would need to compare ECRW builds between stock and my fw to see if there's something in the ECRW update causing a problem

Sorry I forgot the details, do you need my assistant for that?

druby-luke, 2024-09-18:
Hi, is there any update on this? im having the same issue with my CXI3. 

fun thing is, non of my usb c chargers seem to work AFTER removing the screw and installing coreboot. But my OEM Lenovo adapter from my thinkpad with 65 watt seem to work: 

lenovo SA10R16921 , to be exact.

bam80, 2024-09-18:
@druby-luke I reported the issue upstream, and they requested the test with USB-C PD power adapter (vs display with PD output I have). I don't have the adapter currently, so please chime in and report here carefully, with your hardware models, etc.:
https://issuetracker.google.com/issues/222675683#comment3

Also, it would be helpful if you could make your own debug cable for tests (or buy it). I could help with that.

Thanks.

bam80, 2024-09-18:
> I reported the issue upstream

@MrChromebox did you follow the replies? They might provide some insight:
https://issuetracker.google.com/issues/222675683#comment11

MrChromebox, 2024-09-18:
there's nothing new there for well over a year, and the EC should be/stay in RW once it's updated and has transitioned from RO to RW unless there's a full power loss.

On my TEEMO boards here, the box will boot without problem from a USB-PD charger via USB-C. It will not boot when connected via USB-C to a monitor which does USB-PD (~85W).

There's no good way for me to debug this since the debug interface is the SuzyQ cable which connects to the USB-C port, so I can't use that and reproduce the issue simultaneously

bam80, 2024-09-18:
> It will not boot when connected via USB-C to a monitor which does USB-PD

IIRC it worked for you earlier, has something changed?
BTW, what is your USB-PD charger?

MrChromebox, 2024-09-18:
I don't recall it ever working with only the USB-C monitor connection.

the PD brick is a Lenovo 65W one, probably from a Chromebook

bam80, 2024-09-18:
> all 3 CN65s I have here boot via USB-C PD and provide video via USB-C PD

@MrChromebox so these were not booted  from display's PD?

MrChromebox, 2024-09-18:
> > all 3 CN65s I have here boot via USB-C PD and provide video via USB-C PD
> 
> @MrChromebox so these were not booted from display's PD?

I assume I meant that comment as the functions worked independently

bam80, 2024-09-18:
> non of my usb c chargers seem to work AFTER removing the screw and installing coreboot

@druby-luke could you name these chargers?

bam80, 2024-09-18:
> I don't recall it ever working with only the USB-C monitor connection.

@MrChromebox could you test with original FW?
It worked for me:
> With original firmware, I can boot from display's PD with screw in place only. If I remove the screw, get bootloop (need to detach from power before removing the screw, otherwise it can still boot).

druby-luke, 2024-09-23:
> > non of my usb c chargers seem to work AFTER removing the screw and installing coreboot
> 
> @druby-luke could you name these chargers?

sure!

UGreen 100 watt GAN charger 
Ikea 45 watt dual charger 
a random nameless 65 watt charger (that confirmed worked before with OEM firmware)
steamdeck OEM 45 watt adapter
Dell 45 watt XPS13 adapter 
LG screen with a usb C cable that runs 90 watts

bam80, 2024-09-23:
@druby-luke thanks, are you sure 45 W should boot at least?

druby-luke, 2024-09-23:
no I'm not, because the Ugreen did not work I just tried every usb C charger I could find at home haha.

bam80, 2024-09-23:
> a random nameless 65 watt charger (that confirmed worked before with OEM firmware)

@MrChromebox don't you think even without proper debugging equipment, we should be able to bisect the problem?

MrChromebox, 2024-09-23:
> @MrChromebox don't you think even without proper debugging equipment, we should be able to bisect the problem?

the problem seems to exist with the stock firmware and the WP screw removed as well, no? 

Has anyone tried with the screw reinstalled after flashing my firmware?

bam80, 2024-09-23:
> the problem seems to exist with the stock firmware and the WP screw removed as well, no?

In my case that was true, see the OP:

> With original firmware, I can boot from PD with screw in place only. If I remove the screw, get bootloop (need to detach from power before removing the screw, otherwise it can still boot).

  
> Has anyone tried with the screw reinstalled after flashing my firmware?

I tried it both ways (and cold boot too), dead end:
> First of all it still boots from power adapter.
Then, if I swap to PD with WP screw in place, LED blinking red and it doesn't boot.
If I remove the screw, it bootloops.

MrChromebox, 2024-09-23:
I'd have to go digging deep into the PD code in the EC to try and sort this out, and it's not something I'm familiar with at all

bam80, 2024-09-23:
Still, could bisecting give us some low hanging fruits, for start? Looks probable to me.

MrChromebox, 2024-09-23:
bisecting what exactly?

bam80, 2024-09-23:
I would start with building some upstream fw version and test it.
If it works - start to apply your changes, test again and thus bisect.

MrChromebox, 2024-09-23:
there are no changes in coreboot that would affect the EC/PD at all.

The ECRW update is just the tip of the firmware branch.  One could test using Google's latest ECRW update instead of mine and see if that behaves the same way as the stock firmware, that would allow us to have an endpoint to bisect from

bam80, 2024-09-23:
> One could test using Google's latest ECRW update instead of mine

How one would do that, exactly?
Does something prebuilt exist, or we need to build ourselves?
Could you describe the steps in details, or maybe you could test it yourself?

MrChromebox, 2024-09-23:
actually, looks like Google's latest ECRW update for Fizz is the same as mine save for the version string

MrChromebox, 2024-09-23:
here's a test build with Google's ECRW vs mine, just in case it's a compiler issue. If this behaves the same as the public release, then I suspect the issue might be something with depthcharge/vboot

[coreboot_edk2-fizz-mrchromebox_20240923.zip](https://github.com/user-attachments/files/17104856/coreboot_edk2-fizz-mrchromebox_20240923.zip)

bam80, 2024-09-23:
Thank you, unfortunately, I can't test myself at the moment.
I would certainly assist but not now.
So if you could test it yourself that would be awesome.

bam80, 2024-09-24:
@MrChromebox Awesome, it's great to hear the progress!

Some preliminary questions:
> keeping the EC in RO mode allows it to boot via PD from a monitor connection
- How one could reproduce your success test to confirm it? There is still a chance we have mix of the problems here.
- Do you have idea about short-term workaround? Long-term solution?
- Could we adopt TCPMv2 on Fizz and try it there?
https://chromium.googlesource.com/chromiumos/platform/ec/+/HEAD/docs/usb-tcpmv2.md

> It's the RO->RW transition that's causing the reset

It's still not clear to me:
Stock FW still can boot with the WP screw in, why can't we behave the same?

MrChromebox, 2024-09-24:
> * How one could reproduce your success test to confirm it? There is still a chance we have mix of the problems here.

flash a test build which doesn't switch the EC from RO to RW at boot

> * Do you have idea about short-term workaround? Long-term solution?

keep the EC in RO. fix the transition if possible

> * Could we adopt TCPMv2 on Fizz and try it there?
>   https://chromium.googlesource.com/chromiumos/platform/ec/+/HEAD/docs/usb-tcpmv2.md

probably not, Fizz doesn't use a dedicated TCPM chip and backporting the code to Fizz's EC would not be a small undertaking

> > It's the RO->RW transition that's causing the reset
> 
> It's still not clear to me: Stock FW still can boot with the WP screw in, why can't we behave the same?

who knows

bam80, 2024-09-24:
> flash a test build which doesn't switch the EC from RO to RW at boot

If that succeeds, could we incorporate the change into your fork?

> keep the EC in RO

Is RO still a flash memory? Could we flash it by any mean - with a programmer or debug cable?

>  It's still not clear to me: Stock FW still can boot with the WP screw in, why can't we behave the same?

I mean - with the screw in, stock fw does PD contract solely in EC RW, right? So the RO->RW transition doesn't happen. Can we do the same, eliminating the transition?
IIUC, in your test above it happens in RO, but can we do it in RW instead?

MrChromebox, 2024-09-24:
> > flash a test build which doesn't switch the EC from RO to RW at boot
> 
> If that succeeds, could we incorporate the change into your fork?

no, because RO firmware is (much) older, and I can't guarantee stability with a 8 year old RO build
 
> > keep the EC in RO
> 
> Is RO still a flash memory? Could we flash it by any mean - with a programmer or debug cable?

RO is just a flash region on the EC flash chip. It can be flashed like any other. But it's not as easy as RW -- ideally you flash the region that isn't active.

> I mean - with the screw in, stock fw does PD contract solely in EC RW, right? So the RO->RW transition doesn't happen. Can we do the same, eliminating the transition? IIUC, in your test above it [...]

No. The stock EC is always in RW except when the device is booted into recovery mode.

bam80, 2024-09-24:
> RO is just a flash region on the EC flash chip. It can be flashed like any other. But it's not as easy as RW -- ideally you flash the region that isn't active.

Maybe that could be an optional workaround on our side - support flashing RO from the install script. Could we think about it?

> With the stock firmware, EC is always in RW

Anyway, what prevents us from doing the same?

MrChromebox, 2024-09-24:
> > RO is just a flash region on the EC flash chip. It can be flashed like any other. But it's not as easy as RW -- ideally you flash the region that isn't active.
> 
> Maybe that could be an optional workaround on our side - support flashing RO from the install script. Could we think about it?

Absolutely not, it's not safe to do.

> > With the stock firmware, EC is always in RW
> 
> Anyway, what prevents us from doing the same?

we are doing the same thing effectively. Without power, the EC boots in RO, switches to RW (via coreboot), then stays in RW until power lost. I don't have any explanation as to why USB-PD + DP-alt mode  with the stock firmware but not upstream.

bam80, 2024-09-24:
> it's not safe to do.

I guess so, that is why I said _optional_. If user is affected by this issue and has the debug cable, could he flash RO relatively safe?

MrChromebox, 2024-09-24:
> > it's not safe to do.
> 
> I guess so, that is why I said _optional_. If user is affected by this issue and has the debug cable, could he flash RO relatively safe?

I don't have either the tools or a process to direct people on how to do that

bam80, 2024-09-24:
> we are doing the same thing effectively. Without power, the EC boots in RO, switches to RW (via coreboot), then stays in RW until power lost.

- With the WP screw in, when the first time PD contract starts in both cases (our and stock ) - in RW?
- If so, if the switch happens via coreboot, does it mean AP starts initially without the contract?
- Can we use the same EC RW as stock?

MrChromebox, 2024-09-24:
> > we are doing the same thing effectively. Without power, the EC boots in RO, switches to RW (via coreboot), then stays in RW until power lost.
> 
> * With the WP screw in, when the first time PD contract starts in both cases (our and stock ) - in RW?

RO

> * If so, if the switch happens via coreboot, does it mean AP starts initially without the contract?

switch happens in coreboot with upstream, in vboot with stock

> * Can we use the same EC RW as stock?

I already tested that last night, the behavior is the same as with my update.

bam80, 2024-09-24:
>  > With the WP screw in, when the first time PD contract starts in both cases (our and stock ) - in RW?

> RO

The google docks and the answer in the bugtracker above states it's in RW, isn't it?

https://chromium.googlesource.com/chromiumos/platform/ec/+/HEAD/docs/usb-c.md#implementation-considerations

> PD communication is disabled in locked RO images (normal state for customer devices). When the jump from RO to RW happens relatively quickly (e.g. there is not a long memory training step), then there aren't many problems when RW takes over and negotiates higher PD contracts.

https://issuetracker.google.com/issues/222675683#comment5
> PD is disabled in RO for security reason. When WP is disabled, we consider the device is not protected. So, we let the EC negotiate PD in RO. This is not always good because EC-RW doesn't completely take over the PD contract made by EC-RO. If it's not a handled case, EC-RW resets the port.

MrChromebox, 2024-09-24:
it depends on whether power was completely removed from the EC, it can't start in RW from a cold boot

bam80, 2024-09-24:
I thought where EC starts and where PD contract starts are two different things not necessary correlating to each other, and those descriptions are for cold boot as it's not stated the opposite.

MrChromebox, 2024-09-24:
> I thought where EC starts and where PD contract starts are two different things not necessary correlating to each other, and those descriptions are for cold boot as it's not stated the opposite.

having the screw installed does not automatically enable WP -- it just prevents changes to software WP. One would have to examine the EC code to see how exactly it determines if WP is enabled / whether to enable PD for RO

MrChromebox, 2024-09-24:
I have a colleague at Google looking into the bug report / might be able to provide some direction here.
https://issuetracker.google.com/issues/172220407#comment4
would seem to show that the issue can be resolved by not renegotiating. I can carry that change in my builds if I know what it is -- we'll see what I can find out.

aaboagye, 2024-09-24:
One potential solution would be to do the following:

- Add a battery backed RAM (BBRAM) variable to store an index
- Save off the PDO index whenever a PD request message is sent; place this index in the BBRAM variable
- After the PD task starts and checks for an existing explicit contract, you could consult the BBRAM variable for the index and re-request that instead of vSafe5V.

MrChromebox, 2024-09-25:
> One potential solution would be to do the following:
>
> * Add a battery backed RAM (BBRAM) variable to store an index
> * Save off the PDO index whenever a PD request message is sent; place this index [...]

great, let me know when you have it implemented

druby-luke, 2024-09-25:
Small update from me, just received brand new Lenovo 65 watt chargers and they seem to work fine with the current release mrchromebox firmware without the write screw inserted.

Lenovo 4X20M26272 to be exact.

MrChromebox, 2024-09-28:
I've been unsuccessful so far with modifying the PD code, need to have a better understanding of what's going on, so have a USB-PD sniffer on the way (thanks to a friend at Google) which hopefully will allow me to figure out the problem

bam80, 2024-10-06:
> have a USB-PD sniffer on the way (thanks to a friend at Google) which hopefully will allow me to figure out the problem

Thanks for sharing us the details. Did you get the sniffer?

MrChromebox, 2024-10-06:
> Thanks for sharing us the details. Did you get the sniffer?

not yet

druby-luke, 2024-11-17:
Hi! is there perhaps any update on this?

MrChromebox, 2024-11-17:
nope, still waiting to get the USB-PD sniffer to be able to debug further

bam80, 2024-11-19:
> nope, still waiting to get the USB-PD sniffer to be able to debug further

If there is no estimate when it happens, maybe just try to blindly implement the solution suggested by @aaboagye above, meanwhile?
The test is simple: it's either boots or not.

MrChromebox, 2024-11-19:
> > nope, still waiting to get the USB-PD sniffer to be able to debug further
> 
> If there is no estimate when it happens, maybe just try to blindly implement the solution suggested by @aaboagye above, meanwhile? The test is simple: it's either boots or not.

I already tried that without success

MrChromebox, 2025-01-13:
so I've recently been testing running the EC in RO mode, and seems to work without issue on my TEEMO. I have a single USB-C cable connected to the monitor and nothing else. Firmware menus work, windows works, linux works. Only thing I can't do with the RO EC firmware is auto-power on when power is connected regardless of previous state -- that was a RW enhancement I added

bam80, 2025-01-13:
Thanks for the update.
So you can reproduce the issue, but only in RW mode? Did you incorporate any changes to fix it?
Could we develop adoptable solution for other users suffering it?

MrChromebox, 2025-01-13:
@bam80 the USB-PD issue is caused by the RO to RW transition, and so keeping the EC in RO mode mitigates that. The only change is a settings toggle in the UEFI Settings Menu to disable the EC-RW jump.

bam80, 2025-01-13:
> The only change is a settings toggle in the UEFI Settings Menu to disable the EC-RW jump.

I didn't know about that menu (and probably didn't see it, did it appear in the recent FW versions?), could we document it somewhere, as it's not obvious at all it is tied to this issue?

MrChromebox, 2025-01-13:
> I didn't know about that menu (and probably didn't see it, did it appear in the recent FW versions?), could we document it somewhere, as it's not obvious at all it is tied to this issue?

it doesn't exist in released firmware, I literally added it over the weekend just for testing. It appears to resolve the issue here, so I can offer it for testing shortly

[32 more replies]

closed 2025-06-27 by MrChromebox