MrChromebox/firmware #596: Samsung Chromebook Go (XE310XDA) - no touchpad, no camera

kamahell87, 2024-03-19:
I followed the guide flashing the custom firmware on my chromebook and the whole process was very smooth.

When I booted up a Live Fedora, I noticed that the touchpad, the audio and the built-in camera didn't work.
I tried different distros, even Endeavour and Manjaro, with the same result.

I ended up installing Debian 12 and [this script](https://github.com/WeirdTreeThing/chromebook-linux-audio) to fix the audio and at the same time installing a custom kernel.
Now the audio is working just fine, but the touchpad and the camera are still not getting detected by the system.

I double checked the hardware connection and everything well connected and not damaged.

I wonder if it something related to the `firmware-util.sh`  (🤷)
Speaking of the script, I noticed something interesting...
I tried to run it again to see if there was an update to roll out, and there was one! So I installed it and rebooted the machine. When I get back to it, I wanted to satisfy myself by running your script again (expecting to see no updates available) but it showed the same update available. A bug, perhaps?

Thanks for your work!

MrChromebox, 2024-03-20:
please provide cbmem output from a fresh boot, so I can verify that the firmware is detecting the touchpad. Audio and camera are completely OS/driver issues, not a firmware one

kamahell87, 2024-03-20:
Here is the `cbmem -c` output:
```
[NOTE ]  coreboot-4.22-123-g822782a10016-MrChromebox-4.22.2-5-g822782a1001 Tue Feb 13 17:47:11 UTC 2024 x86_32 bootblock starting (log level: 7)...
[DEBUG]  CPU: Intel(R) Celeron(R) N4500 @ 1.10GHz
[DEBUG]  CPU: ID 906c0, Jasperlake A0, ucode: 24000024
[DEBUG]  CPU: AES supported, TXT NOT supported, VT supported
[DEBUG]  MCH: device id 4e22 (rev 00) is Jasperlake SKU4-1
[DEBUG]  PCH: device id 4d87 (rev 01) is Jasperlake Super
[DEBUG]  IGD: device id 4e55 (rev 01) is Jasperlake GT4
[DEBUG]  FMAP: Found "FLASH" version 1.1 at 0xc04000.
[DEBUG]  FMAP: base = 0xff000000 size = 0x1000000 #areas = 8
[DEBUG]  FMAP: area COREBOOT found @ c04200 (4177408 bytes)
[INFO ]  CBFS: mcache @0xfef31a00 built for 19 files, used 0x3e0 of 0x4000 bytes
[INFO ]  CBFS: Found 'fallback/romstage' @0x80 size 0x10b60 in mcache @0xfef31a2c
[DEBUG]  BS: bootblock times (exec / console): total (unknown) / 1 ms

[NOTE ]  coreboot-4.22-123-g822782a10016-MrChromebox-4.22.2-5-g822782a1001 Tue Feb 13 17:47:11 UTC 2024 x86_32 romstage starting (log level: 7)...
[DEBUG]  pm1_sts: 8100 pm1_en: 0000 pm1_cnt: 00001c00
[DEBUG]  gpe0_sts[0]: 00000010 gpe0_en[0]: 00000000
[DEBUG]  gpe0_sts[1]: 00000000 gpe0_en[1]: 00000000
[DEBUG]  gpe0_sts[2]: 00000000 gpe0_en[2]: 00000000
[DEBUG]  gpe0_sts[3]: 00080000 gpe0_en[3]: 00000000
[DEBUG]  TCO_STS:   0000 0001
[DEBUG]  GEN_PMCON: d0801038 00002200
[DEBUG]  GBLRST_CAUSE: 00000040 00000000
[DEBUG]  prev_sleep_state 5
[DEBUG]  FMAP: area COREBOOT found @ c04200 (4177408 bytes)
[INFO ]  CBFS: Found 'fspm.bin' @0x68dc0 size 0x79000 in mcache @0xfef31c98
[DEBUG]  FMAP: area RW_MRC_CACHE found @ bb0000 (65536 bytes)
[DEBUG]  SPD INDEX = 0
[INFO ]  CBFS: Found 'spd.bin' @0x368c0 size 0x200 in mcache @0xfef31bd8
[INFO ]  SPD: module type is LPDDR4X
[INFO ]  SPD: module part number is                     
[INFO ]  SPD: banks 8, ranks 1, rows 16, columns 10, density 16384 Mb
[INFO ]  SPD: device width 16 bits, bus width 32 bits
[INFO ]  SPD: module size is 4096 MB (per channel)
[INFO ]  meminit_channels: DRAM half-populated
[DEBUG]  CBMEM:
[DEBUG]  IMD: root @ 0x76fff000 254 entries.
[DEBUG]  IMD: root @ 0x76ffec00 62 entries.
[DEBUG]  FMAP: area RO_VPD found @ c00000 (16384 bytes)
[WARN ]  init_vpd_rdev: No RW_VPD FMAP section.
[DEBUG]  External stage cache:
[DEBUG]  IMD: root @ 0x7b3ff000 254 entries.
[DEBUG]  IMD: root @ 0x7b3fec00 62 entries.
[DEBUG]  FMAP: area RW_MRC_CACHE found @ bb0000 (65536 bytes)
[DEBUG]  MRC: Checking cached data update for 'RW_MRC_CACHE'.
[INFO ]  SF: Detected 00 0000 with sector size 0x1000, total 0x1000000
[DEBUG]  MRC: 'RW_MRC_CACHE' does not need update.
[DEBUG]  cse_lite: Number of partitions = 3
[DEBUG]  cse_lite: Current partition = RW
[DEBUG]  cse_lite: Next partition = RW
[DEBUG]  cse_lite: Flags = 0x7
[DEBUG]  cse_lite: RO version = 13.50.11.1304 (Status=0x0, Start=0x1000, End=0xebfff)
[DEBUG]  cse_lite: RW version = 13.50.11.1304 (Status=0x0, Start=0x140000, End=0x2c2fff)
[DEBUG]  1 DIMMs found
[DEBUG]  SMM Memory Map
[DEBUG]  SMRAM       : 0x7b000000 0x800000
[DEBUG]   Subregion 0: 0x7b000000 0x200000
[DEBUG]   Subregion 1: 0x7b200000 0x200000
[DEBUG]   Subregion 2: 0x7b400000 0x400000
[DEBUG]  top_of_ram = 0x77000000
[DEBUG]  Normal boot
[INFO ]  CBFS: Found 'fallback/postcar' @0x118740 size 0x56d4 in mcache @0xfef31d0c
[DEBUG]  Loading module at 0x76c1f000 with entry 0x76c1f031. filesize: 0x5348 memsize: 0xb698
[DEBUG]  Processing 211 relocs. Offset value of 0x74c1f000
[DEBUG]  BS: romstage times (exec / console): total (unknown) / 2 ms

[NOTE ]  coreboot-4.22-123-g822782a10016-MrChromebox-4.22.2-5-g822782a1001 Tue Feb 13 17:47:11 UTC 2024 x86_32 postcar starting (log level: 7)...
[DEBUG]  Normal boot
[... 2280 lines cut ...]
[DEBUG]  Fixed MTRRs   : Enabled
[DEBUG]  Variable MTRRs: Enabled

[INFO ]  CBFS: Found 'fallback/payload' @0x11de80 size 0x13ee4b in mcache @0x76c2d350
[DEBUG]  Checking segment from ROM address 0xffd220ac
[DEBUG]  Checking segment from ROM address 0xffd220c8
[DEBUG]  Loading segment from ROM address 0xffd220ac
[DEBUG]    code (compression=1)
[DEBUG]    New segment dstaddr 0x00800000 memsize 0x1000000 srcaddr 0xffd220e4 filesize 0x13ee13
[DEBUG]  Loading Segment: addr: 0x00800000 memsz: 0x0000000001000000 filesz: 0x000000000013ee13
[DEBUG]  using LZMA
[DEBUG]  Loading segment from ROM address 0xffd220c8
[DEBUG]    Entry Point 0x0080168c
[DEBUG]  BS: BS_PAYLOAD_LOAD run times (exec / console): 710 / 0 ms
[DEBUG]  Finalizing chipset.
[DEBUG]  apm_control: Finalizing SMM.
[DEBUG]  APMC done.
[DEBUG]  BS: BS_PAYLOAD_LOAD exit times (exec / console): 1 / 0 ms
[INFO ]  HECI: Sending End-of-Post
[INFO ]  CSE: EOP requested action: continue boot
[DEBUG]  BS: BS_PAYLOAD_BOOT entry times (exec / console): 7 / 0 ms
[DEBUG]  mp_park_aps done after 0 msecs.
[DEBUG]  Jumping to boot code at 0x0080168c(0x76b59000)

```

MrChromebox, 2024-03-20:
this is a SASUKE board? cbmem shows no touchpad detected. Also EC update failing, which isn't good.

kamahell87, 2024-03-20:
The board is SASUKETTE. The touchpad was definitely working with ChromeOS

kamahell87, 2024-03-20:
> looks like SASUKETTE was not properly supported, but instead using the SASUKE image. I've gone ahead and added support for SASUKETTE, but in order to get the correct image, you will need to go back to ChromeOS first so the device is detected properly via the script. Please LMK if it's not detected properly

So, let me see if I got it right.
I need to use use the stock firmware I backed up using your script, then install ChromeOS back, and finally do the whole procedure back again to install the Coreboot and then Linux.

MrChromebox, 2024-03-20:
correct. There's no good way to force your device to go from sasuke to sasukette without manually flashing, and I don't want to do that over github

kamahell87, 2024-03-20:
Perfect. I'll try to do the whole procedure before the end of the week.
I'll keep you posted!

kamahell87, 2024-03-21:
I went though the whole process and the touchpad is recognised as is should! The built in camera is working as well. The only issue, is the audio that doesn't come out but is being recognised in the settings. 
Thank you very much for your help and for your work!

MrChromebox, 2024-03-21:
audio isn't going to work OOTB, will need to use the chrultrabook script for that. But that's an OS issue not a firmware one. Glad the others are working now though :)

closed 2024-03-21 by MrChromebox