MrChromebox/firmware #597: TPM on c1030 doesn't work on Linux

Integral-Tech, 2024-03-19:
When I tried to enroll the TPM in order to use it to unlock the LUKS volume, it returned an error:

```
ERROR:esys:src/tss2-esys/api/Esys_TestParms.c:96:Esys_TestParms() Esys Finish ErrorCode (0x000b0143)

Failes to create TPM2 context: State not recoverable
```

Model: HP Elite c1030 chromebook

MrChromebox, 2024-03-19:
the firmware does it's part to set up the TPM, you likely need to reset/clear ownership etc. But check with cbmem

Integral-Tech, 2024-03-19:
> the firmware does it's part to set up the TPM, you likely need to reset/clear ownership etc. But check with cbmem

How can I check with cbmem?

Integral-Tech, 2024-04-17:
> the firmware does it's part to set up the TPM, you likely need to reset/clear ownership etc. But check with cbmem

https://github.com/systemd/systemd/issues/31925#issuecomment-2034895544

MrChromebox, 2024-04-17:
> > the firmware does it's part to set up the TPM, you likely need to reset/clear ownership etc. But check with cbmem
> 
> [systemd/systemd#31925 (comment)](https://github.com/systemd/systemd/issues/31925#issuecomment-2034895544)

there's nothing I can do, the firmware on the CR50 is signed by Google, and they chose to implement a subset of the TPM 2.0 spec

closed 2024-04-17 by Integral-Tech