MrChromebox/firmware #605: Device becomes unresponsive upon Ctrl+L

retroflexivity, 2024-03-31:
Device: Acer cp713-3w (VOXEL)

Installed the RW_LEGACY firmware, but can't legacy boot. On the developer mode screen, upon pressing ctrl+l, the device freezes infinitely on the same screen. The USB stick, if plugged in, also seems to get disconnected at that point (stops glowing).

The same happens if I choose 'select alternative bootloader' > 'tianocore'. Booting from internal storage is working.

would include some logs, but don't know how to get them, sorry

MrChromebox, 2024-04-01:
there's not much I can do to debug RW_LEGACY not working on a given device, since I'd need to have debug stock firmware, which isn't available. Sometimes the stock firmware is just broken for RWL

retroflexivity, 2024-04-01:
I wonder if there's anything I can do. and whether there's a chance a fix is coming

MrChromebox, 2024-04-01:
no to both unfortunately. Only thing I could do is rebuild the RW_LEGACY payload and hope that it was an edk2 issue that's since been fixed

retroflexivity, 2024-04-02:
would be great, thank you

I suppose this fact doesn't increase the chance the device will get bricked if I install UEFI?

retroflexivity, 2024-04-02:
great. pls tell me when you rebuild the payload, i will try it and go the UEFI way if it fails

retroflexivity, 2024-04-04:
just uefied it, all smooth. thank you for the firmware mrchromebox

closed 2024-04-04 by retroflexivity