MrChromebox/firmware #606: Asus Chromebook CX1400CNA no me deja reinstalar Chrome Os

Anibalmania, 2024-03-31:
Hola, me llamo Aníbal, gusto en saludarles. Queria saber si pueden ayudarme con una Asus Chromebook CX1400CNA, le cambie el firmware para instalar windows 10 y no pude instalar el sistema operativo porque no me dejo, apareció una pantalla azul y se reiniciaba, abri Linux e hice restauración del firmware que descarga de la pagina con el script pero ahora no me deja instalar en ChromeOS nuevamente, me da error, y ya no puedo volver a instalar el UEFI, practicamente quedado como inservible y queria saber si habia algun modo de recuperarla, muchas gracias 😌

MrChromebox, 2024-04-01:
try performing the recovery with the internal battery disconnected

MrChromebox, 2024-04-16:
closing due to lack of follow up

closed 2024-04-16 by MrChromebox