MrChromebox/firmware #607: Installed (UEFI) Full ROM on my Lenovo C340-11 but rebooting just leads to a black screen

Hexseas, 2024-04-07:
Searched a bit around for how to fix this but am left confused. Seems A Suzy Qable is the closest thing to a solution. But I would rather have another method to fix my problem. It doesn’t recognise the device I used to install the thingo earlier in the process. And the local and verified image selections just lead to a black screen. So any solutions?

MrChromebox, 2024-04-07:
I don't understand the problem

> Seems A Suzy Qable is the closest thing to a solution.  But I would rather have another method to fix my problem.

huh?

>  It doesn’t recognise the device I used to install the thingo earlier in the process

what is "it"? thingo?

> And the local and verified image selections just lead to a black screen

those are options  when doing what?

Hexseas, 2024-04-07:
Boot splash screen appears then it goes to a black screen.

MrChromebox, 2024-04-07:
"I tried different options" doesn't tell me anything. You flashed the UEFI firmware, so you need to install a UEFI-bootable OS from USB. Have you done that?

Hexseas, 2024-04-08:
So I am supposed to install UEFI-bootable on a usb?

PhantomTimed, 2024-04-08:
I might be completely misunderstanding, but your saying so far you... are trying to boot into your devices stock firmware? That's all I can get from this.

Hexseas, 2024-04-08:
I basically followed this video (https://youtu.be/z6oyqrrXTLM?si=kiuSkRpa68ULI6RY) until 13:23 but i got a black screen. Stock firmware? No, I installed the UEFI full rom, I don’t think I’m trying to get into chromeos. I just need to figure out why it’s doenst let me and just gives me a black screen.

MrChromebox, 2024-04-08:
>  I just need to figure out why it’s doenst let me and just gives me a black screen.

if you can't communicate what you've done, what you're trying to do, and what the device is doing, then we can't help you.

PhantomTimed, 2024-04-08:
> I basically followed this video (https://youtu.be/z6oyqrrXTLM?si=kiuSkRpa68ULI6RY) until 13:23 but i got a black screen. Stock firmware? No, I installed the UEFI full rom, I don’t think I’m trying [...]

What OS does the USB have on it? If none, what files does the USB have on it? (The one your trying to boot)

Hexseas, 2024-04-08:
My problem is similar to the closed #586 but I don’t have any os installed locally. And I ran the ChromeOS Firmware Utility Script, with wp disabled, selected menu option 2. Now it boots up to a black screen. Literally all I did.

PhantomTimed, 2024-04-09:
None of us here are quite understanding... you seem to be attempting to boot into ChromeOS with UEFI Full ROM firmware as no OS has been installed locally. If you want to dual-boot, use RW_LEGACY if available. Please put the firmware backup onto your PC, then flash an OS using balenaEtcher. Summary: Set it up like you set up your host PC.

Hexseas, 2024-04-09:
I will try what you said initially but yea any way of booting I am aware of (just powering it on, default boot, reboot) results in a black screen.

PhantomTimed, 2024-04-09:
That means that ChromeOS is still installed. You must flash a UEFI-compatible OS onto a usb (keep your firmware backup!) then replace the ChromeOS install with the new OS.

PhantomTimed, 2024-04-09:
This issue may probably be closed now as it is not a firmware issue, just a user mistake (which is okay!)

Hexseas, 2024-04-09:
Any suggestions for an os?

PhantomTimed, 2024-04-09:
I would recommend tiny11 for Windows 11, or GalliumOS as a Linux distro

MrChromebox, 2024-04-09:
I pushed a fix to the script that will prevent the ChromeOS remnants from being booted after flashing the UEFI firmware

closed 2024-04-10 by Hexseas