MrChromebox/firmware #608: 32 MB UMA_FRAME_BUFFER allocated to Vega 8 iGPU by default and can't change it.

Tu2l, 2024-04-07:
Hi @MrChromebox ,
First of all, thank you very much for your great contribution to the Chromebook community. 
A few days ago, I flashed the firmware by using your scripts and installed Windows 11 and tested it out. During my tests, I found out that the memory allocated to the internal GPU, which is Vega 8 in my case, is only 32 MB, and there is no option to change it from the core boot's advance configuration (the advance configuration is always empty). This amount of GPU memory is not enough to even play HD or FHD videos on YouTube; however, in Linux (OpenSuse), the amount of GPU memory shown is around 256MB. 

My Chromebook config:

Board: morpheus hfan
CPU: AMD Ryzen 5 3500C
GPU: AMD VEGA 8 iGPU
Memory: 16GB
Storage: 256GB m.2 ssd

MrChromebox, 2024-04-07:
modern integrated GPUs which use shared system memory don't need a large fixed amount, the drivers can dynamically allocate it any time it's needed. 32MB is more than enough for FHD or even UHD video, if you do the pixel math.

Tu2l, 2024-04-07:
I'm not sure if it is actually dynamically allocating any memory to iGPU, in my case i was never able to play any yt videos above 360p. In paper math 32MB video memory might be enough but in reality with all these bg tasks, i dont think 32MB of video memory can play UHD videos. From my experience i have seen AMD iGPUs reserves some portion of RAM to use as video memory it never changes during runtime of OS, only way to change this memory allocation is to change uma buffer size in bios settings.

MrChromebox, 2024-04-07:
if you're having issues playing videos above 360p, then you have other issues, not a UMA size issue.

Tu2l, 2024-04-07:
Is it possible this might be related to some incompatibility with windows ? 

In linux the video memory size was 256mb same as chromeos and video play issue occurred only once, after restart video playback started normally.

MrChromebox, 2024-04-07:
coolstar / the chrultrabook project handles Windows compatibility, I just build the firmware

MrChromebox, 2024-04-16:
the 4.22.3 release ups the UMA to 128MB

closed 2024-04-16 by MrChromebox

MrChromebox, 2024-06-18:
> Hi @MrChromebox, I just saw this commit in your coreboot repo [MrChromebox/coreboot@89de0a9](https://github.com/MrChromebox/coreboot/commit/89de0a92d1f2c4d52e590db7b3df00c7ea780f8a) , will this commit impact the UMA size for Lenovo Thinkpad Yoga C13 (Morphius) ?

not at all, Kahlee is the StoneyRidge baseboard, completely separate from Picasso boards like Morphius. Those are at 128MB for now.

Tu2l, 2024-06-21:
Thanks for the clarification, just wanted to know if it is possible to increase the UMA size for morphius broad and build the rom/frimware locally ?

MrChromebox, 2024-06-21:
> Thanks for the clarification, just wanted to know if it is possible to increase the UMA size for morphius broad and build the rom/frimware locally ?

no, if it were I would have provided instructions to do so.

On PCO, the UMA size (and mode) is set as an option  inside the compiled APCB binary. The source and tools to compile that binary are not available outside of AMD (and NDA partners).

Tu2l, 2024-07-17:
Hi @MrChromebox today I've tested linux (opensuse) using RW legacy bios/alternate bootloader and it seems the uma buffer size or the video memory is still 64MB only.

MrChromebox, 2024-07-17:
> Hi @MrChromebox today I've tested linux (opensuse) using RW legacy bios/alternate bootloader and it seems the uma buffer size or the video memory is still 64MB only.

I'd expect so, the RW_LEGACY firmware is just that -- a bootloader -- and cannot change anything set by the stock firmware. You need to use the UEFI Full ROM firmware to get that.

Tu2l, 2024-08-05:
Hi @MrChromebox, I'm curious to know if it is possible to access coreboot advance settings using **nvramtool** of coreboot.

I encountered a git repo which does the same but its for a specific hardware, but if it is possible to access advance settings using **nvramtool** then I would like to build a similar tool for chromebook users so that they can have full access to their bios/firmware. 

Git repo - https://github.com/JaGoLi/corevantage

MrChromebox, 2024-08-05:
@Tu2l these options have to be set up for each individual board, and the nvram options have to be enabled. Neither is done for any of the ChromeOS devices I support. So in short, there are no options to configure.

Tu2l, 2024-08-05:
Got it, thanks for the reply.

Tu2l, 2024-08-10:
Hi @MrChromebox, One request from my side is to increase the UMA size for Zork boards to 256MB or 512MB, as these boards are quite capable of handling intermediate-level development work.

MrChromebox, 2024-08-10:
what benefit would that actually provide? 512MB UMA on a 4GB device is a big chunk

Tu2l, 2024-08-10:
I do actually have 16 GB of memory, and I want to test out some graphic intensive tools like godot game engine and other similar tools and some light gaming.

MrChromebox, 2024-08-10:
there's no way for me to set it based on the system memory size, and I don't want to be offering multiple builds for the same device. I wasn't even aware of any 16GB zork devices being released

Tu2l, 2024-08-10:
Yeah, I can understand you; bumping up the UMA size might negatively impact devices with low system memory, but isn't it possible to go with the standard 256 MB of UMA for all the devices?

MrChromebox, 2024-08-12:
I'll bump it to 256MB for the 2408 release and see if anyone complains

Tu2l, 2024-08-12:
Thank you very much.

Tu2l, 2024-08-26:
Hi @MrChromebox, Just for your information, yesterday I flashed the full rom using your script, and it bricked my device. I'm not sure what went wrong here; in the past I flashed it using the same script, and it was successful because of 64MB of VRAM I reverted, but now the same script bricked it. Maybe the recent Rom update caused this. 

OpenSuse Tumbleweed was used to flash the full rom (note: I already had RW_legacy installed).

Right now I'm trying to build a SuzyQ cable or Debugging cable to flash the stock firmware.

MrChromebox, 2024-08-26:
this is on MORPHIUS? Did you try a hard reset (Refresh+Power simultaneously)? I tested the ROM here on mine and no issues, am puzzled that others are running into problems

Tu2l, 2024-08-26:
Yes, its on morphious.

Tu2l, 2024-09-10:
Hi @MrChromebox, thank you very much for 2408 release. As i can see in release notes now, it is possible to adjust UMA size while compiling the rom but unfortunately I'm not able to find any documentation regarding it.
Is there any specific documentation for compiling the ROM with UMA-specific configuration?

MrChromebox, 2024-09-10:
> Hi @MrChromebox, thank you very much for 2408 release. As i can see in release notes now, it is possible to adjust UMA size while compiling the rom but unfortunately I'm not able to find any [...]

the size is selected in the device's config file under configs/pco/config.{board name}.uefi

valid values are 64/128/256/512MB. 512 is what I'm using for the release.

Tu2l, 2024-09-10:
It's already set to the max so no need to compile it again i guess. Once again thank you very much for your dedication.

Tu2l, 2025-06-14:
Hi @MrChromebox,

I just wanted to know if it would be possible to add two additional UMA options for the Google Zork mainboard.

Currently, I observe the following supported UMA sizes in the firmware:

> ./src/mainboard/google/zork/uma_128.apcb
> ./src/mainboard/google/zork/uma_256.apcb
> ./src/mainboard/google/zork/uma_512.apcb (default)
> ./src/mainboard/google/zork/uma_64.apcb

My request is to include options for 1 GiB and 2 GiB UMA sizes. It would be ideal if the default selection of 512 MiB could remain unchanged. This would greatly benefit users like myself who wish to compile custom firmware with larger UMA allocations for specific use cases or improved performance.

Thank you for considering this request.

Regards,