MrChromebox/firmware #610: Ubuntu and Firmware install/update script incorrectly identify Hardware Model

mex20, 2024-04-09:
When I try to use the Firmware install/update script to update my Asus C433T, I get the error: ```The script does not currently have a firmware file for your device (RAMMUS); Cannot continue.``` 

I am running Ubuntu 22.04.4 LTS and I am on a firmware version from a couple of years ago when I initially flashed the MrChromebox firmware to this machine. The initial script that I used a few years ago correctly identified the Hardware Model, but now it is not identifying this machine correctly and therefore cannot update to the latest firmware version. I have attached screenshots to show the issue and the About page in Ubuntu settings.

[image]
[image]

MrChromebox, 2024-04-09:
just pushed a fix, give it a try in a min or so. you'll need to select which device you have when prompted (should be SHYVANA)

mex20, 2024-04-09:
Thank you very much @MrChromebox! This fix worked for me and I was able to upgrade my firmware.

closed 2024-04-09 by mex20