MrChromebox/firmware #611: Wireless keyboard not detected until after boot

stangosnell, 2024-04-09:
Asus CN60, flashed with Seabios years ago, has been running fine, but now won't accept keyboard input until after boot, and sometimes the receiver has to be removed/reinserted.  I can't get to the boot control screen, it just boots directly to the OS, Debian Sid.  I want to boot from a USB drive to install Debian Stable, but I cannot get it to accept any keyboard input until the login screen appears.  The keyboard works fine after boot, and it works from power on on another computer side by side with the CN60.  I've tried using a USB hub and directly into all the onboard ports.

MrChromebox, 2024-04-09:
sounds like a SeaBIOS bug, and I really can't support firmware that's 5+ years old. Update to the latest UEFI firmware and then install your OS. Remove the internal SSD if needed to force it to boot from an external source.

stangosnell, 2024-04-10:
I've tried removing the SSD, and booting from a Ventoy drive.  It boots to the .iso selection screen, but I can't select one to boot because the keyboard doesn't work.  I'm unsure how to install the firmware from a running OS, but I will investigate that.

MrChromebox, 2024-04-10:
don't use a ventoy drive, use a Ubuntu (eg) ISO written directly to USB via Etcher.

> I'm unsure how to install the firmware from a running OS

https://mrchromebox.tech/#fwscript

stangosnell, 2024-04-10:
Running the firmware script from the OS gives the following:
`Downloading supporting files...

Downloading required tools...

Getting device/system info...

Flashrom is unable to read current firmware; cannot continue:

flashrom 1.4.0-devel on Linux 6.7.9-amd64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0x7f73d000.
Found chipset "Intel Lynx Point LP Mainstream".
This chipset is marked as untested. If you are using an up-to-date version
of flashrom *and* were (not) able to successfully update your firmware with it,
then please email a report to flashrom@flashrom.org including a verbose (-V) log.
Thank you!
Enabling flash write... Error accessing ICH RCRB, 0x4000 bytes at 0x00000000fed1c000
/dev/mem mmap failed: Operation not permitted
FAILED!
FATAL ERROR!
Error: Programmer initialization failed.

You may need to add 'iomem=relaxed' to your kernel parameters,
or trying running from a Live USB with a more permissive kernel (eg, Ubuntu 23.04+).

See https://www.flashrom.org/FAQ for more info.
`
Booting from a live CD is unsuccessful, because of the lack of keyboard input to select the boot type.

stangosnell, 2024-04-10:
I should have tried this from the start.  My apologies for not reading the error message closely.  Setting the kernel parameters and updating grub allowed the script to run successfully, and solved the keyboard issue.  It's all good now.

MrChromebox, 2024-04-10:
awesome, glad the issue is resolved :)