MrChromebox/firmware #614: Lenovo Flex 5 13ITL6 Back Screen

forlhynet, 2024-04-18:
[log.txt](https://github.com/MrChromebox/firmware/files/15028476/log.txt)
after flashing full uefi firmware,device got black screen(no any display),the attachment is ec log,this device is a test device,hwid is "lindar test"

MrChromebox, 2024-04-18:
the ec log isn't going to tell me anything, need the AP/CPU log (ttyUSB1) from a serial-enabled build. You have a SuzyQ to flash and capture the log?

forlhynet, 2024-04-19:
[tty1.txt](https://github.com/MrChromebox/firmware/files/15042354/tty1.txt)
thanks for your reply. this is the log on ttyUSB1

MrChromebox, 2024-04-19:
log doesn't show a full boot, can you try again? looks like it is trying to switch to the RW CSE and keeps rebooting

forlhynet, 2024-04-19:
I tried several times,and the console stops at "ChromeEC SW Sync: EC_RW is up to date" always

forlhynet, 2024-04-19:
is it a problem that the device memory is 4g, but the log shows 8g

MrChromebox, 2024-04-19:
> is it a problem that the device memory is 4g, but the log shows 8g

can you get me a cbmem log from stock firmware/ChromeOS?

forlhynet, 2024-04-19:
how to geta cbmem log?i have the stock rom file

MrChromebox, 2024-04-19:
boot ChromeOS, go to chrome://system and then bios_log IIRC

forlhynet, 2024-04-19:
[Uploading stock.txt…]()
stock rom logs

forlhynet, 2024-04-19:
[stock.txt](https://github.com/MrChromebox/firmware/files/15043738/stock.txt)
stock rom logs

forlhynet, 2024-04-20:
it seems that the device shut down after calling "google_chromeec_reboot(EC_REBOOT_JUMP_RW, 0)"

MrChromebox, 2024-04-20:
I'll build an image with the stock ChromeOS ECRW update and see if that resolves the issue

MrChromebox, 2024-04-20:
please flash and provide the cbmem log from this firmware, uses Google's latest ECRW update

[coreboot_edk2-lindar-mrchromebox_20240420.zip](https://github.com/MrChromebox/firmware/files/15048701/coreboot_edk2-lindar-mrchromebox_20240420.zip)

forlhynet, 2024-04-21:
still the same situation, referring to the ec log, it seems that the EC recognized the wrong battery status and triggered force shutdown,i built a firmware to disable swsync and successfully booted it into efisell,but the ec firmware is an old version
[image]
[image]

forlhynet, 2024-04-21:
[log11.txt](https://github.com/MrChromebox/firmware/files/15050458/log11.txt)

MrChromebox, 2024-04-21:
EC sync isn't the issue, it's the jump to RW. But if it's not booting with Google's ECRW, then there's something else that's the problem. the RO EC firmware is very old and buggy.

forlhynet, 2024-04-22:
Can you compile an ECRW with "intel_x86_wait_power_up_ok()" removed for me?thanks

MrChromebox, 2024-04-22:
not easily, I'd need to set up a new ChromeOS chroot to be able to build TGL/volteer ECRW images

forlhynet, 2024-04-22:
Okay, I'll try compiling it myself. Thanks for your help

forlhynet, 2024-04-23:
Is there any way to upgrade  RO ec.bin?

MrChromebox, 2024-04-23:
sure, if you're feeling really F'in brave. The whole point of the RO-RW split is to prevent you from getting screwed by a bad RW update. I would revert to ChromeOS, allowing it to update the ECRW itself, then flash a coreboot image without ecrw/ecrw.hash. Then it will only jump to ECRW without updating, same as ChromeOS

closed 2024-05-18 by forlhynet

MrChromebox, 2024-05-19:
@forlhynet what was the resolution here?